lvgl/lv_binding_rust #5: Button callbacks

justinmoon, 2020-05-15:
First off, it seems like event-handling has been implemented in `lvgl_sys`, but hasn't been implemented in `lvgl` yet. Is that accurate?

I'm trying to add a button example and [I'm stuck trying to type the callback to `lvgl_sys::lvgl_obj_set_event_cb`](https://github.com/justinmoon/lvgl-rs/blob/50105eb3dcb2cbcb8dfb4cf0427a199d861aa058/examples/button/src/main.rs#L42)

It seems like this callback needs to be casted to `lvgl_sys::lv_event_cb_t` but we can't use `as` since it's not a primitive type. Is this something where we'd want to add a `From` trait to the `Button` object in `objx.rs`?

justinmoon, 2020-05-15:
It seems like `lvgl_sys` wants the callback to be an `unsafe extern "C"` function, which doesn't really make sense if I'm providing this callback from Rust ...

rafaelcaricio, 2020-05-15:
Hi @justinmoon,

It is flattering that you have interest in using lvgl-rs. I just wanted to tell you that this project is in the very early stages of development. I would really appreciate help of others, like you, to make push this project forward. Most things are still missing and I'm still pondering about how to handle the global state dependency in lvgl library rely.

justinmoon, 2020-05-15:
No problem. I'll try to help out wherever I can.

rafaelcaricio, 2020-05-15:
In general, I'm trying to follow the recommendations from those articles:
- http://adventures.michaelfbryan.com/posts/how-not-to-riir/
- https://kornel.ski/rust-sys-crate

justinmoon, 2020-05-15:
Thanks for sharing. 

I'm trying to figure out how to get these event callbacks working. [This post](https://users.rust-lang.org/t/callback-based-c-ffi/26583/5) seems to explain roughly what is needed, but still working on it

justinmoon, 2020-05-15:
On a related note, how were you able to get a button to look like this? 
[image]

Mine look like this
[image]

rafaelcaricio, 2020-05-27:
@justinmoon that screenshot was in an earlier version of the demo app, https://github.com/rafaelcaricio/lvgl-rs/blob/7bf1ac0cae3b1867b4687efaf1ccbd9e321df54b/examples/demo/src/main.rs I was using SDL directly.

rafaelcaricio, 2020-05-31:
Hey @justinmoon , I just implemented a way to add event callbacks. So you can do things when buttons are clicked, hovered, etc. Please look the example:
```rust
let mut button = Button::new(&mut screen);
button.on_event(|this, event| {
    if let lvgl::Event::Clicked = event {
        println!("Clicked!");
    }
});
```
https://github.com/rafaelcaricio/lvgl-rs/blob/master/examples/bar.rs#L48-L52

rafaelcaricio, 2020-05-31:
I will close this ticket now, as this is supported. If you have other questions. Please open other issue.

closed 2020-05-31 by rafaelcaricio