lvgl/lv_binding_rust #7: std build-dependencies

sajattack, 2020-05-16:
I'm trying to test this out on an embedded target and it's pulling in std due to the build dependencies. How is it supposed to work?

sajattack, 2020-05-16:
Maybe the problem is I'm using xargo? Not sure.

rafaelcaricio, 2020-05-16:
@sajattack I'm happy you are interested in lvgl-rs. I have to tell you that this library is not ready. It is in very early stage of development. Barely any code was wrote to it so far. I would need help of contributors like yourself to find out how to solve issues like the one you are having. I have personally not tested this crate in a device yet, though I do not use any [`std`](https://github.com/rafaelcaricio/lvgl-rs/blob/master/lvgl/src/lib.rs#L1) in the code.

sajattack, 2020-05-16:
https://github.com/rust-lang/cargo/issues/4866

sajattack, 2020-05-16:
The resolution is to add -Zfeatures=itarget,build_dep when compiling.

closed 2020-05-16 by sajattack