lvgl/lv_binding_rust #8: Tracking issue for running on embedded platforms

sajattack, 2020-05-17:
So currently, this crate has only been tested in the embedded-graphics simulator. I have encountered two problems compiling for an embedded target:
- [x] std build-dependencies (resolved by compiling with -Zfeatures=build_dep)
- [ ] unhelpful bindgen error (see below)

```
thread 'main' panicked at 'libclang error; possible causes include:
- Invalid flag syntax
- Unrecognized flags
- Invalid flag arguments
- File I/O errors
- Host vs. target architecture mismatch
If you encounter an error missing from this list, please file an issue or a PR!', /home/paul/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.53.2/src/ir/context.rs:567:13
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

make: *** [Makefile:9: release] Error 101
```

I'm happy to contribute to get this to work but I'm a little bit stumped. I thought I should at least document it here in case anyone else can figure it out. I look forward to using this crate in the future.

rafaelcaricio, 2020-05-17:
Thank you @sajattack ! I'm also not having much time lately. But I will try to get into this when I have some time.

rafaelcaricio, 2020-05-30:
@sajattack It should be working now. I made a basic mistake of forgetting to include `#![no_std]` to the lvgl-sys crate. Please try again and let me know.

rafaelcaricio, 2020-05-30:
Btw, we still need the flag `-Zfeatures=build_dep` to build.
That should be not needed after https://github.com/rust-lang/cargo/issues/7915 is fixed.

sajattack, 2020-05-30:
Needs alloc?
```
error: no global memory allocator found but one is required; link to std or add `#[global_allocator]` to a static item that implements the GlobalAlloc trait.
```

rafaelcaricio, 2020-05-30:
Yes, we need alloc. You can find how to setup an allocator here: https://github.com/rafaelcaricio/pinetime-rtfm/blob/master/pinetime-rtfm/src/main.rs#L37 
I will publish later a simplified example of how to do this.

sajattack, 2020-05-30:
Ok, just not very `#![no_std]` friendly to have alloc.

rafaelcaricio, 2020-05-30:
Well, LittlevGL do allocate memory dynamically. So there is not really a workaround for that.

rafaelcaricio, 2020-05-30:
@sajattack I have updated the README with the points we discussed in this thread so others are aware. https://github.com/rafaelcaricio/lvgl-rs#building-for-embedded-environments

Please let me know if we can close this issue. Also, you are welcome to amend the README.md with any other info you think would be important to know. I really appreciate you help and interest in this project.

sajattack, 2020-05-30:
I'm testing now. Currently getting a display full of static, is there a method I need to call to flush lvgl to the display?

rafaelcaricio, 2020-05-30:
You need to call those two methods:
```rust
ui.tick_inc(Duration::from_millis(5));
ui.task_handler();
```
So LittlevGL will load the framebuffer and flush to the display.
https://docs.lvgl.io/v7/en/html/porting/project.html#initialization

sajattack, 2020-05-30:
Sorry, it's not working for me and I don't have time to debug.

rafaelcaricio, 2020-05-30:
Ok, no worries. I will update the readme with some step-by-step so it is easier for everyone (including me).

rafaelcaricio, 2020-05-31:
As the main concern in this ticket is solved, I will close. Please open a new ticket for further questions.

closed 2020-05-31 by rafaelcaricio

rafaelcaricio, 2020-06-03:
@sajattack About your concern on the dependency on `alloc`, maybe this idea would make it flexible enough: https://github.com/rafaelcaricio/lvgl-rs/issues/16