lvgl/lv_binding_rust #15: Create WebAssembly demo app

rafaelcaricio, 2020-06-02:
Create a WebAssembly demo app using the [web simulator](https://github.com/rahul-thakoor/embedded-graphics-web-simulator) of `embedded_graphics` library. It would be a great way to show the possibilities of `lvgl-rs` by compiling to WASM. The example can also be a quick start for many people.

FloVanGH, 2020-09-17:
Does anyone managed to run it on WebAssembly? When I try to build it I get the following error:

```
_core/../lv_font/../lv_misc/lv_area.h:17:10: fatal error: 'string.h' file not found
  cargo:warning=#include <string.h>
```

rafaelcaricio, 2020-09-17:
Hey @FloVanGH , I haven't tried yet. But I'd imagine that to compile to WebAssembly we would need to tweak our `build.rs` to handle cases like this one.

FloVanGH, 2020-09-17:
If you can adjust the `build.rs` I can contribute an example 🙂.

FloVanGH, 2020-09-18:
I want to add an lvgl example to my crate: https://codeberg.org/morph/canvas-display. I have also a working desktop example for an other crate https://codeberg.org/morph/raqote-display.

rafaelcaricio, 2020-09-18:
@FloVanGH I can work on updating the `build.rs`, but I'm pretty tight on free time right now so I cannot compromise when I can do it. I hope to get some time to work on lvgl-rs soon.

rafaelcaricio, 2020-09-18:
Awesome work, btw! :)

FloVanGH, 2020-09-18:
> @FloVanGH I can work on updating the `build.rs`, but I'm pretty tight on free time right now so I cannot compromise when I can do it. I hope to get some time to work on lvgl-rs soon.

Ok I see. It's not urgent. But I hope you'll find some time. And I hope you can continue your work, it's really great 🙂.

FloVanGH, 2020-09-18:
> Awesome work, btw! :)

Thank you 🙂

rafaelcaricio, 2020-09-21:
I've tried that this weekend and didn't work yet.  I will continue trying to compile for the `wasm32-unknown-emscripten` target, that one may work (https://github.com/rustwasm/team/issues/291).

FloVanGH, 2020-09-21:
Hey, ok thank you for your effort 🙂.

rafaelcaricio, 2020-09-21:
@FloVanGH after many hours of trial and error, it's possible to compile lvgl-rs to WASM after those changes https://github.com/rafaelcaricio/lvgl-rs/pull/31

I have an example here: https://github.com/rafaelcaricio/lvgl-rs-wasm

FloVanGH, 2020-09-21:
@rafaelcaricio that's awesome :-) . Thank you. I will create an example for `canvas-display`.

rafaelcaricio, 2020-09-21:
@FloVanGH No problem! Let me know if it works for you in the browser. As I said, even though I managed to make the example project compile successfully, it panics when running in the browser for some strange reason.

rafaelcaricio, 2020-09-21:
[image]
This is the error I get... :/

FloVanGH, 2020-09-22:
Ok thank you. I will check it.

rafaelcaricio, 2020-09-22:
@FloVanGH  I managed to make it work! 🥳  The example project is running here: https://rafaelcaricio.github.io/lvgl-rs-wasm/

Example code: https://github.com/rafaelcaricio/lvgl-rs-wasm

FloVanGH, 2020-09-22:
🎉

closed 2021-06-03 by rafaelcaricio