lvgl/lv_binding_rust #21: undefined symbol: strcmp

sajattack, 2020-06-13:
Hey, it's me again. Trying to experiment with this some more but ran into a linker error. 
```
  = note: rust-lld: error: undefined symbol: strcmp
          >>> referenced by lv_img_cache.c:91 (/home/paul/.cargo/git/checkouts/lvgl-rs-27f20f23afade818/335261e/lvgl-sys/vendor/lvgl/src/lv_draw/lv_img_cache.c:91)
          >>>               lv_img_cache.o:(_lv_img_cache_open) in archive /home/paul/projects/rust/st7735-lcd-examples/target/thumbv7em-none-eabihf/debug/deps/liblvgl_sys-46c501e1cf67e4f2.rlib
          >>> referenced by lv_img_decoder.c:327 (/home/paul/.cargo/git/checkouts/lvgl-rs-27f20f23afade818/335261e/lvgl-sys/vendor/lvgl/src/lv_draw/lv_img_decoder.c:327)
          >>>               lv_img_decoder.o:(lv_img_decoder_built_in_open) in archive /home/paul/projects/rust/st7735-lcd-examples/target/thumbv7em-none-eabihf/debug/deps/liblvgl_sys-46c501e1cf67e4f2.rlib
```

rafaelcaricio, 2020-06-13:
Hey, nice to hear from you! I'm changing the code pretty heavily right now. And actually testing in a device apart from the simulator. If you want to test the latest, please use the version in this branch `generate-code`.

rafaelcaricio, 2020-06-13:
This is the fix https://github.com/rafaelcaricio/lvgl-rs/pull/20/files#diff-94577d69799c72315f5d7adeb8a37595R70 for your issue.

sajattack, 2020-06-13:
Ok thanks.

sajattack, 2020-06-13:
The version of proc-macro2 you're using conflicts with the one in cortex-m-rt.

closed 2020-06-13 by sajattack

rafaelcaricio, 2020-06-13:
That dependency is a leftover, can be removed. Please pull again the branch. Anyways, it is a work in progress.. I'm working on `lvgl-rs` right now. :nerd_face: 

I will actually change sometime soon how the binding code is generated in that branch.