lvgl/lv_binding_rust #28: Cannot implement From<Color> for new types

Spadi0, 2020-08-25:
The `UI` object requires an `embedded_graphics` display to draw to, and the provided display needs to use a `PixelColor` that also implements `From<Color>`. However, the `Color` type does not export any fields or functions that provide either a `lv_color_t`, or RGBA fields.

This makes it impossible to convert from the `Color` type, as no color data is provided; this means that only displays that use `Rgb565` or `Rgb888` work (as they have crate-provided implementations).

Either the `raw` field needs to be public:
https://github.com/rafaelcaricio/lvgl-rs/blob/d442ab99cc015d4a3d814de7d188b5792a19aea7/lvgl/src/support.rs#L18
or a function/functions that return RGBA data are needed.

rafaelcaricio, 2020-08-27:
Check out this https://github.com/rafaelcaricio/lvgl-rs/pull/29

I'm not sure the `alpha` information is provided by LVGL. 🤔

rafaelcaricio, 2020-08-27:
You can reopen if you find this is not useful. Also feel free to send pull requests. Unfortunately, I am short on free time lately to work on `lvgl-rs`. I hope to be back at it in a few weeks.

PR #29 (2020-08-27): Expose RGB values from Color
Fixes #28
closed 2020-08-27 by rafaelcaricio