lvgl/lv_binding_rust #30: Cannot build on Redox

FloVanGH, 2020-09-18:
I know it is not the main purpose of this crate, but it would be really cool if I can built it on Redox.

I've tried it but I get the following error:

```shell
Compiling lvgl-sys v0.4.0
The following warnings were emitted during compilation:

warning: couldn't execute `llvm-config --prefix` (error: No such file or directory (os error 2))
warning: set the LLVM_CONFIG_PATH environment variable to the full path to a valid `llvm-config` executable (including the executable itself)

error: failed to run custom build command for `lvgl-sys v0.4.0`

Caused by:
  process didn't exit successfully: `/home/flovanrg/Projects/redox/cookbook/recipes/lvgl/build/target/debug/build/lvgl-sys-7891e4f7ab4aa66a/build-script-build` (signal: 6, SIGABRT: process abort signal)
``` 

Maybe one of you can help me. I will also ask in the Redox community for help.

I want to run the following example I've created:

https://codeberg.org/morph/raqote-display/src/branch/main/examples/lvgl

closed 2020-09-18 by FloVanGH

rafaelcaricio, 2020-09-18:
Did it work in the end?

FloVanGH, 2020-09-18:
Not yet. I have some feedback from the Redox community.

> Try using this one instead in makefile -export-dynamic

> Try bumping [this](https://github.com/servo/libexpat/blob/master/expat/CMakeLists.txt#L6) to at least 3.4 (as per [CMP0065](https://cmake.org/cmake/help/v3.6/policy/CMP0065.html). It solved my -rdynamic issues in the past with other ports.

I will try it and hope it will work.