lvgl/lv_binding_rust #33: LVGL-rs on Raspberry Pi

MGlolenstine, 2020-10-16:
Will this library work on RPi CM3+ with or without X?

I'm trying to work on an "embedded" GUI, but Linux and it's framebuffer isn't mentioned in the list of "supported displays", so I'm unsure. Lvgl itself [can be run on Raspberry Pi](http://www.vk3erw.com/index.php/16-software/63-raspberry-pi-official-7-touchscreen-and-littlevgl), but I'm wondering what's the status of it in `lvgl-rs`.

rafaelcaricio, 2020-10-16:
I didn't try to run in Raspberry Pi, but I see no reason why it wouldn't work. As long as you have access to the framebuffer, you can create your own [embedded-graphics driver](https://docs.rs/embedded-graphics/0.6.2/embedded_graphics/#implementing-embedded_graphics-in-a-driver) for it. I run `lvgl-rs` on nRF52832 platform which are way less powerful than a Raspberry Pi.

MGlolenstine, 2020-10-16:
Oh, so framebuffer driver needs to be implemented before we can use it.
Then that's great news, I'll look into doing that and make it work.
The question wasn't about performance, but about ability to use framebuffers and draw onto screens detected by OS and not baremetal.

Thanks!

rafaelcaricio, 2020-10-16:
Cool, let me know how it goes. 🙂

nia-e, 2023-03-02:
Once #64 gets merged native framebuffer support should be added if you're still interested, @MGlolenstine

MGlolenstine, 2023-03-02:
I'm not in the immediate need of this, as I've had to switch projects due to unrelated issues, but I'd love to have it accessible if I return to the project someday :)