lvgl/lv_binding_rust #34: Linker Error

vinaychandra, 2020-10-19:
When trying to use the library in `no-std` context, I am getting the following linker error

I just added `lvgl` as the cargo dependency and tried to create an example with it.

Should lvgl's native libraries be included manually?

```
> Executing task: make <

DEP_LV_CONFIG_PATH=`pwd` cargo build -Zfeatures=itarget,build_dep --target ./triplets/x86_64-thunderstorm.json -p thunderstorm-kernel
   Compiling thunderstorm-kernel v0.0.2 (/Users/vinay/Coding/Thunderstorm)
warning: variable does not need to be mutable
   --> src/main.rs:120:9
    |
120 |     let mut screen = ui.scr_act().unwrap();
    |         ----^^^^^^
    |         |
    |         help: remove this `mut`
    |
    = note: `#[warn(unused_mut)]` on by default

warning: unused `core::result::Result` that must be used
  --> src/main.rs:71:5
   |
71 |     load_graphics();
   |     ^^^^^^^^^^^^^^^^
   |
   = note: `#[warn(unused_must_use)]` on by default
   = note: this `Result` may be an `Err` variant, which should be handled

error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "--script=./triplets/thunderstorm_x86_64.ld" "--eh-frame-hdr" "-L" "/Users/vinay/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/x86_64-thunderstorm/lib" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.111hhfeqn0nygrs7.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.11b5kflgqro1ohm1.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.11q1gro77dgb2fr4.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.13adwmfy2i2prw81.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.13qrr933h5ocms2b.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.145iud1vm6fj957c.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.154ohh92086zxve5.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.171jxo9g5r1g2wzp.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1ebpyxqf67yq4iv1.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1fl14pooj97hkjd7.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1fpyqpry9s3p3olt.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1iama4ptk4v9lpui.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1isbq9bdjme231qn.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1k7z69ysdb8v4eri.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1k98m2fjr97hlid7.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1m28soqrzfdhmfa8.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1msrsjsvehh88t6u.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1ndalu13ukec0hu2.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1nr6i543yvdfuusq.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1od0eyn0io3w60wn.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1pd8lddrvq80ynfe.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1rhx0c609uck1c9y.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1s141wti9wmqnqbb.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1s95ck1fou7rfpt5.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1tk7jsyt072n4kx7.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1ujm9q3h9x12ts1p.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1vskg8go8d55f953.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1w2myeh4knvbjqq7.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1wixcb6oqpoddwnw.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1xbvc2fwqks64jg5.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1xe2fhan045nwfas.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1y4judcbzpw94wge.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1y6hduyywwyyewiy.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1yapiemz9f6e78x3.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.1zwfp6po59gh1w0f.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.200q5gs9zqvfpbut.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.21mlp96lunyylsnf.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.223doi3q8e4sa39u.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.232f9j3kri6nbyby.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.238u2kxlkp5natqv.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.23bjecrwfjddn7yx.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.23yzsr7ikksv2tnw.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.23zo7vjp115lilt9.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.24la3mxj6u6hgbbg.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.24p2agax6b3xyb8k.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.2545i2sib9bx5633.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.25l8p86j98vn214u.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.25wz1tvkvi4hf5bq.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.26vbhs9uo8fipq43.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.29jilgzvf0i2rje7.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.2al7794ikygkmx0s.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.2aluf4rff290frz5.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.2b9whd8lx1bqy3d.rcgu.o" "/Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/thunderstorm_kernel-b85206a8885cc465.2c2kfwb1vtkq0w0x.rcgu.o" "/User [... cut ...]  in archive /Users/vinay/Coding/Thunderstorm/target/x86_64-thunderstorm/debug/deps/liblvgl-75c9211469d9ecb7.rlib
          

error: aborting due to previous error; 2 warnings emitted

error: could not compile `thunderstorm-kernel`

To learn more, run the command again with --verbose.
make: *** [target/x86_64-thunderstorm/debug/thunderstorm-kernel] Error 101
The terminal process "/bin/zsh '-c', 'make'" terminated with exit code: 2.

Terminal will be reused by tasks, press any key to close it.
```

rafaelcaricio, 2020-10-19:
That's an interesting error. The lvgl native src is vendored into the `lvgl-rs` crate. What version of the Rust compiler are you using?

vinaychandra, 2020-10-19:
This is the latest rust nightly 

My suspicion is that the cc compilation is not being done in a cross platform way. My host is mac and target is not. Mac's symbols can contain extra underscore in symbol names. May be bindgen and cc are being run differently?

Note that i can run examples fine. And examples are not cross compiled.

nia-e, 2023-03-28:
Hi! There have been a lot of updates since this issue was opened, including some pretty comprehensive changes to the C code compilation ever since #67; does this issue still occur?

vinaychandra, 2023-03-28:
I unfortunately haven't used this in a long time. I cannot answer this question currently. Feel free to close this if you think this is not worth anymore. If i ever get back to this, i will reopen the issue.

nia-e, 2023-03-28:
That's fair. I unfortunately don't have access to a macOS machine to test this presently; I'll spin up a VM tomorrow and see if I can replicate the issue and likely close if not. Regardless, feel free to reopen or open a new issue if you have any questions or follow-ups!