lvgl/lv_binding_rust #35: Running the demos on macOS fails, when sdl2 is missing

rwbr, 2020-11-16:
I'm on macOS Big Sur 11.0.1. When I try to run one of the demos I get the following error message:

```
...

"-framework" "Security" "-lSDL2" "-lSystem" "-lresolv" "-lc" "-lm"
  = note: ld: library not found for -lSDL2
          clang: error: linker command failed with exit code 1 (use -v to see invocation)
          

error: aborting due to previous error

error: could not compile `demo`.

To learn more, run the command again with --verbose.

```

To make the samples work I manually have to install sdl2 first by issuing the following command:

`brew install sdl2`

After that, the samples work on macOS too. Please update the README accordingly. Thank's.

rafaelcaricio, 2020-11-16:
@rwbr Thank you for the suggestion. Do you want to open a PR for this? I would be happy to accept it. If not, that is fina as well. I will update when I get some time.

rwbr, 2020-11-16:
Sure! I will update the documentation and open a PR when done.

PR #36 (2020-11-17): Update README: Added a hint to install SDL2 before running the demos on macOS
At least in my case it was necessary to install libsdl (SDL2) before I was able to compile the demos on my Mac. I added a hint to the README in order to help other developers that stumble upon that problem.

Fixes #35
closed 2020-11-17 by rafaelcaricio