lvgl/lv_binding_rust #37: How to get involved?

shymega, 2020-11-20:
Hi,

I'm using this crate for some embedded firmware I'm working on, and I'm aware that some features are missing.

I wanted to ask if it were possible to have tracker issues to give those who want to help, a good place to start coding on?

Thanks!

rafaelcaricio, 2020-11-25:
Hi @shymega,

Thank you for the interest in helping! I've been quite busy lately and could not work as much as I would've liked in lvgl-rs in the last few months. Your help is more than welcome. I will post here later a longer description on what I am thinking and how/where you can jump in to help.

Cheers!

shymega, 2020-11-26:
Hey. Looking forward to it! I had a bit of trouble compiling, but figured out I needed to enable LVGL in the `lv_conf.h` in the end.. :)

rafaelcaricio, 2020-12-06:
Hi @shymega,

I finally got around to write back to you.

So the status of the project is that it works for the most basic usage, like drawing widgets to the screen and setting the basic options. We don't have a way to create custom styles yet, we can set some properties directly into the widgets though. Customization of fonts are also not possible unless using `unsafe`. My ideal scenario in general is to make it possible for people to use this crate without ever needing to write any `unsafe` code.

When it comes to project structure. We have three crates:
- lvgl: Main crate intended for final users, people who want to write LVGL apps in Rust.
- lvgl-sys: The bindgen generated interface to the C headers.
- lvgl-codegen: The code generation for common/simpler ABI calls. In this crate I try to identify some common code patterns when creating the bindings and generalize in a way that Rust binding code can be generated. Since LVGL has many methods/functions, this helps a lot on improving the bindings coverage.

All those crates comes together in the [`lvgl` crate](https://github.com/rafaelcaricio/lvgl-rs/blob/81cb1bee4c2107e553caa5016860c1e9c674ae00/lvgl/Cargo.toml#L24-L25). We use the `lvgl-codegen` to read [the output of the bindgen](https://github.com/rafaelcaricio/lvgl-rs/blob/81cb1bee4c2107e553caa5016860c1e9c674ae00/lvgl/build.rs#L13) and from the function signatures, we try to generate the [binding code](https://github.com/rafaelcaricio/lvgl-rs/blob/81cb1bee4c2107e553caa5016860c1e9c674ae00/lvgl/build.rs#L13-L32). The `lvgl-codegen` has mapping to what to do depending on the arguments of the functions. In this test you can see how it works:

https://github.com/rafaelcaricio/lvgl-rs/blob/81cb1bee4c2107e553caa5016860c1e9c674ae00/lvgl-codegen/src/lib.rs#L556-L589

From the bindgen code to the final [LVGL Rust binding](https://github.com/rafaelcaricio/lvgl-rs/blob/81cb1bee4c2107e553caa5016860c1e9c674ae00/lvgl-codegen/src/lib.rs#L205-L213). The basic object structure is defined using this macro https://github.com/rafaelcaricio/lvgl-rs/blob/81cb1bee4c2107e553caa5016860c1e9c674ae00/lvgl/src/lv_core/obj.rs#L120 in this file you can also understand the basic structure of all Widget objects in the Rust LVGL. If you want to define something extra and specific for some Widget, we can add here https://github.com/rafaelcaricio/lvgl-rs/tree/81cb1bee4c2107e553caa5016860c1e9c674ae00/lvgl/src/widgets.

The basic setup of display and all the bootstrap code necessary for LVGL is in this file https://github.com/rafaelcaricio/lvgl-rs/blob/81cb1bee4c2107e553caa5016860c1e9c674ae00/lvgl/src/ui.rs

What I would like to do next? Increase the coverage of the generated code, ideally we could have most of the bindings auto-generated. So contributions to the `lvgl-codegen` are the most wanted right now. I would like to find a way to generate all those enums:

https://github.com/rafaelcaricio/lvgl-rs/blob/81cb1bee4c2107e553caa5016860c1e9c674ae00/lvgl/src/widgets/label.rs#L13-L20

You can see here that this could well be generated.

```rust
#[derive(Debug, Copy, Clone, PartialEq)]
#[repr(u8)]
pub enum LabelAlign {
    Left = lvgl_sys::LV_LABEL_ALIGN_LEFT as u8,
    Center = lvgl_sys::LV_LABEL_ALIGN_CENTER as u8,
    Right = lvgl_sys::LV_LABEL_ALIGN_RIGHT as u8,
    Auto = lvgl_sys::LV_LABEL_ALIGN_AUTO as u8,
}
```

Find everything that matches `LV_LABEL_` which belongs to the Label Object type. Then find everything that belongs to `LABEL_ALING_` and then create the arms `LEFT`, `CENTER`, `RIGHT`, and `AUTO`. Then split, convert to PascalCase and add all together in an Rust `enum`.

Let me know if you get the idea. You're free to implement anything, but just keep in mind if and how it could be generalized in the `lvgl-codegen`, that is the preferred place to contribute right now. Also you can see the code in `lvgl-codegen` is all in one file, that is because I am still focusing on making it work first before starting to split in multiple files and think about a nice code architecture. Since it is a completely new approach for me, I don't have a clear view yet on how to make the code "look nice" at this stage.

rafaelcaricio, 2020-12-06:
Sorry if my last comment was confusing. You can of course ask my anything and I will answer what I know. :)

shymega, 2021-03-14:
Yeah, I have a repo with GNU Guile bindings, and the single file output is a total pain. Gotcha with porting. Should I just make PRs for each feature/enhancement - like adding a new enum to `LabelAlign` -  for example? The `lvgl-codegen` request looks pretty simple. What would be great is an automatically generated list of entries remaining - but I don't think its possible! 😆

shymega, 2021-03-14:
Also, would you be able to add the 'Help Wanted' label to this issue? Apologies for the late reply, too!

rafaelcaricio, 2021-03-15:
@shymega yes, you can sent small PRs. That would be great actually :)

shymega, 2021-03-15:
Alright, I'll take a look. I moved to JetBrains CLion recently, which I'm finding excellent for Rust development -- so, hopefully, I'll be more productive! Note: I am also pursuing employment, so if I do find a job, my time may be more limited, although I'll try to work on this when I can. The firmware I'm working on relies on LVGL for display support, so it's definitely in my interests.

rafaelcaricio, 2021-03-15:
I use CLion as well, it's great. 😄 No rush at all, just have fun... I also work in this project in my free time whenever I feel like. There are no deadlines here 🙈 Good luck in your job hunting! 👍🏽