lvgl/lv_binding_rust #39: Error when running examples

Geometrically, 2021-02-10:
Hi, 
I tried running the examples with the command in the README, however I am getting this error.

Am I doing anything wrong?
```
error[E0432]: unresolved import `syn::export`
 --> lvgl-codegen/src/lib.rs:9:10
  |
9 | use syn::export::ToTokens;
  |          ^^^^^^ could not find `export` in `syn`

error[E0599]: no method named `to_token_stream` found for struct `std::boxed::Box<Pat>` in the current scope
   --> lvgl-codegen/src/lib.rs:251:26
    |
251 |         Self::new(fa.pat.to_token_stream().to_string(), fa.ty.into())
    |                          ^^^^^^^^^^^^^^^ method not found in `std::boxed::Box<Pat>`
    | 
   ::: /home/geometrically/.cargo/registry/src/github.com-1ecc6299db9ec823/quote-1.0.8/src/to_tokens.rs:61:8
    |
61  |     fn to_token_stream(&self) -> TokenStream {
    |        ---------------
    |        |
    |        the method is available for `std::boxed::Box<std::boxed::Box<Pat>>` here
    |        the method is available for `Rc<std::boxed::Box<Pat>>` here
    |
    = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
    |
1   | use quote::ToTokens;
    |

error[E0599]: no method named `to_token_stream` found for struct `std::boxed::Box<syn::Type>` in the current scope
   --> lvgl-codegen/src/lib.rs:320:34
    |
320 |             literal_name: r_type.to_token_stream().to_string(),
    |                                  ^^^^^^^^^^^^^^^ method not found in `std::boxed::Box<syn::Type>`
    | 
   ::: /home/geometrically/.cargo/registry/src/github.com-1ecc6299db9ec823/quote-1.0.8/src/to_tokens.rs:61:8
    |
61  |     fn to_token_stream(&self) -> TokenStream {
    |        ---------------
    |        |
    |        the method is available for `std::boxed::Box<std::boxed::Box<syn::Type>>` here
    |        the method is available for `Rc<std::boxed::Box<syn::Type>>` here
    |
    = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
    |
1   | use quote::ToTokens;
    |
```

rafaelcaricio, 2021-02-15:
Fixed by #40

rafaelcaricio, 2021-02-15:
Thank you for fixing this issue. I haven't experienced that myself.

closed 2021-02-15 by rafaelcaricio