lvgl/lv_binding_rust #53: cross compiling

rtwfroody, 2022-01-13:
I'm trying to cross compile the example, and am having no luck.

Using `cross`:
```
tnewsome@compy-linux:~/projects/fahrenheit/lvgl-rs$ cat Cross.toml 
[build.env]
passthrough = [ "DEP_LV_CONFIG_PATH" ]
tnewsome@compy-linux:~/projects/fahrenheit/lvgl-rs$ DEP_LV_CONFIG_PATH=`pwd`/examples/include cross build --example demo --target=armv7-unknown-linux-gnueabihf --features="alloc"
warning: field is never read: `r_type`
   --> lvgl-codegen/src/lib.rs:306:5
    |
306 |     r_type: Option<Box<syn::Type>>,
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: `#[warn(dead_code)]` on by default

warning: `lvgl-codegen` (lib) generated 1 warning
   Compiling lvgl-sys v0.5.2 (/project/lvgl-sys)
error: failed to run custom build command for `lvgl-sys v0.5.2 (/project/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/target/debug/build/lvgl-sys-c9bd983769ee75ba/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at 'Directory /home/tnewsome/projects/fahrenheit/lvgl-rs/examples/include referenced by DEP_LV_CONFIG_PATH needs to exist', lvgl-sys/build.rs:32:13
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed
```

I assume there is some way to set DEP_LV_CONFIG_PATH to a value that works inside the container, but I've had no luck in figuring out the solution.

Using `--target`:
```
tnewsome@compy-linux:~/projects/fahrenheit/lvgl-rs$ DEP_LV_CONFIG_PATH=`pwd`/examples/include cargo build --target armv7-unknown-linux-gnueabihf --example demo --features=alloc
warning: field is never read: `r_type`
   --> lvgl-codegen/src/lib.rs:306:5
    |
306 |     r_type: Option<Box<syn::Type>>,
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: `#[warn(dead_code)]` on by default

warning: `lvgl-codegen` (lib) generated 1 warning
warning: 1 Build compiler: cc
warning: 1 Build compiler: arm-linux-gnueabihf-gcc
   Compiling lvgl v0.5.2 (/home/tnewsome/projects/fahrenheit/lvgl-rs/lvgl)
error: linking with `cc` failed: exit status: 1
  |
  = note: "cc" "/home/tnewsome/projects/fahrenheit/lvgl-rs/target/armv7-unknown-linux-gnueabihf/debug/examples/demo-2fe78188483d1610.121qmq4pggluyri0.rcgu.o" ...
...
  = note: /usr/bin/ld: /home/tnewsome/projects/fahrenheit/lvgl-rs/target/armv7-unknown-linux-gnueabihf/debug/examples/demo-2fe78188483d1610.121qmq4pggluyri0.rcgu.o: relocations in generic ELF (EM: 40)
...
          collect2: error: ld returned 1 exit status
          

error: could not compile `lvgl` due to previous error
```

The problem there seems to be that it's using `cc` and `ld` instead of the appropriate cross compiler and linker. I added a warning in lvgl-sys/build.rs. You can see that in the first invocation the compiler is set to `cc`, while in the second invocation it is set to `arm-linux-gnueabihf-gcc` which looks correct.

rafaelcaricio, 2022-01-27:
Could you contextualize me here on what is `cross`?

rtwfroody, 2022-01-27:
I was referring to https://github.com/cross-rs/cross, version 0.2.1.

rafaelcaricio, 2022-01-27:
Does this path exists inside the container: `/home/tnewsome/projects/fahrenheit/lvgl-rs/examples/include` ? I don't think so. I believe here you will need to add the path that works for the container in the `DEP_LV_CONFIG_PATH ` env var.

rtwfroody, 2022-02-02:
The path definitely does not exist. I was hoping somebody else had already cross-compiled this library so I wouldn't have to learn too many details about `cross` (or about using `--target`) but it sounds like nobody has yet. I might spend some more time digging into this later.

omarandlorraine, 2022-06-28:
@rtwfroody Did you figure it out?

rtwfroody, 2022-06-30:
> @rtwfroody Did you figure it out?

Nope. I've been slowly investigating a few options, but haven't found something that works yet. My current hope is that I can just compile LVGL for my target myself, and use regular rust infrastructure to call into the C library directly, without the benefit of this abstraction layer.

sven, 2022-12-19:
It's a bit late and maybe some things changed but I managed to compile (not tested yet) the demo with cross:

```shell
cargo install cross --git https://github.com/cross-rs/cross
git clone https://github.com/rafaelcaricio/lvgl-rs.git
cd lvgl-rs
git submodule update --init
```

Create a file named `Cross.toml` in the root of `lvgl-rs` with the following content:

```toml
[target.armv7-unknown-linux-gnueabihf]
pre-build = [ 
    "ln -fs /usr/share/zoneinfo/Europe/Berlin /etc/localtime",
    "dpkg --add-architecture armhf",
    "apt-get update",
    "DEBIAN_FRONTEND=noninteractive apt-get install --assume-yes tzdata",
    "dpkg-reconfigure --frontend noninteractive tzdata",
    "apt-get --assume-yes install clang libclang",
    "apt-get --assume-yes install libsdl2-dev:armhf"
]

[build.env]
passthrough = ["DEP_LV_CONFIG_PATH"]
```

After that the example compiles with the command:

```shell
DEP_LV_CONFIG_PATH=`pwd`/examples/include cross build --target=armv7-unknown-linux-gnueabihf --example demo --features="alloc"
```

The binary is located in the path: `target/armv7-unknown-linux-gnueabihf/debug/examples/demo`

nia-e, 2023-03-02:
@rtwfroody is this still occuring on latest git master? Note that there have been breaking changes between it since this issue was opened.

rtwfroody, 2023-03-05:
I've pretty much abandoned trying this approach. I just pulled the latest from git, and even building locally using the supplied instructions isn't working:
```
tnewsome@compy-linux:~/projects/lvgl-rs$ DEP_LV_CONFIG_PATH=`pwd`/examples/include cargo run --example demo --features="alloc"
   Compiling lvgl-sys v0.5.2 (/home/tnewsome/projects/lvgl-rs/lvgl-sys)
error: failed to run custom build command for `lvgl-sys v0.5.2 (/home/tnewsome/projects/lvgl-rs/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/home/tnewsome/projects/lvgl-rs/target/debug/build/lvgl-sys-62d86d66a7864cf0/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=/home/tnewsome/projects/lvgl-rs/examples/include/lv_conf.h

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', lvgl-sys/build.rs:137:39
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

nia-e, 2023-03-05:
Did you initialize submodules before building?

rtwfroody, 2023-03-05:
Thanks. I had not initialized the submodules. Having done that, I get the exact same failures as when I opened this issue.
Building with `cross` fails because DEP_LV_CONFIG_PATH isn't available inside the docker cross sets up.
Building with --target fails because it's trying to link with `cc` instead of the cross-compiling binary.

nia-e, 2023-03-05:
Maybe explicitly setting a linker in the Cargo `config.toml` could fix this? See https://github.com/nia-e/mxboot/blob/main/.cargo/config.toml (this was on a slightly older version of `lvgl-rs`, namely commit 6d4de8b, but setting the linker should fix things). Note that we're about to merge *more* API-breaking changes as part of the upgrade to LVGL 8.3

sven, 2023-03-05:
Did you try the `Cross.toml` I posted here https://github.com/rafaelcaricio/lvgl-rs/issues/53#issuecomment-1358353789? I tested the steps against the latest `master` branch and it still compiles.

rtwfroody, 2023-03-08:
@sven That still fails:
```
tnewsome@compy-linux:~/projects/lvgl-rs$ cat Cross.toml 
[target.armv7-unknown-linux-gnueabihf]
pre-build = [ 
    "ln -fs /usr/share/zoneinfo/Europe/Berlin /etc/localtime",
    "dpkg --add-architecture armhf",
    "apt-get update",
    "DEBIAN_FRONTEND=noninteractive apt-get install --assume-yes tzdata",
    "dpkg-reconfigure --frontend noninteractive tzdata",
    "apt-get --assume-yes install clang libclang",
    "apt-get --assume-yes install libsdl2-dev:armhf"
]

[build.env]
passthrough = ["DEP_LV_CONFIG_PATH"]
tnewsome@compy-linux:~/projects/lvgl-rs$ DEP_LV_CONFIG_PATH=`pwd`/examples/include cross build --target=armv7-unknown-linux-gnueabihf --example demo --features="alloc"
   Compiling lvgl-sys v0.5.2 (/project/lvgl-sys)
error: failed to run custom build command for `lvgl-sys v0.5.2 (/project/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/target/debug/build/lvgl-sys-62d86d66a7864cf0/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at 'Directory /home/tnewsome/projects/lvgl-rs/examples/include referenced by DEP_LV_CONFIG_PATH needs to exist', lvgl-sys/build.rs:36:13
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to run custom build command for `lvgl-sys v0.5.2 (/project/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/target/debug/build/lvgl-sys-62d86d66a7864cf0/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at 'Directory /home/tnewsome/projects/lvgl-rs/examples/include referenced by DEP_LV_CONFIG_PATH needs to exist', lvgl-sys/build.rs:36:13
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
tnewsome@compy-linux:~/projects/lvgl-rs$ 
```

I really appreciate that you're all trying to help, but I'm not trying to make this work anymore. If you need something that only I can do then I can help, but a few days ago I literally `git clone`d again to test these ideas. I don't think there's anything unique about my environment.

rtwfroody, 2023-03-24:
Cross from `https://github.com/cross-rs/cross` gives me a docker error: `unknown flag: --progress`

If I just run with cross 0.2.5 (simply `cargo install cross` without specifying the git repo) I get the following error:
```
...
   Compiling lvgl-sys v0.5.2 (/project/lvgl-sys)
error: failed to run custom build command for `lvgl-sys v0.5.2 (/project/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/target/debug/build/lvgl-sys-08f3269db868f84c/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at 'Directory /home/tnewsome/projects/tmp/lvgl-rs/examples/include referenced by DEP_LV_CONFIG_PATH needs to exist', lvgl-sys/build.rs:63:13
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to run custom build command for `lvgl-sys v0.5.2 (/project/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/target/debug/build/lvgl-sys-08f3269db868f84c/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at 'Directory /home/tnewsome/projects/tmp/lvgl-rs/examples/include referenced by DEP_LV_CONFIG_PATH needs to exist', lvgl-sys/build.rs:63:13
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Which sounds like the same problem with `cross` I've been running into all along.

I used Cargo.toml from https://github.com/rafaelcaricio/lvgl-rs/issues/53#issuecomment-1358353789

I spent some time trying to hack unsupported flags out of mainline `cross` but just ran into `--build-arg`, which seems important enough that I shouldn't just remove it. I guess my docker (part of Ubuntu 22.04) is from 2020, so maybe just too old. I'll try upgrading it when I have a chance.

rtwfroody, 2023-03-25:
OK, I worked past the docker flags issue by running `cross` with `CROSS_CONTAINER_ENGINE_NO_BUILDKIT=1`.
Now I get:
```
  cargo:warning=In file included from /usr/include/SDL2/SDL_stdinc.h:31,
  cargo:warning=                 from /usr/include/SDL2/SDL_main.h:25,
  cargo:warning=                 from /usr/include/SDL2/SDL.h:32,
  cargo:warning=                 from /home/tnewsome/projects/tmp/lvgl-rs/lvgl-sys/vendor/lv_drivers/sdl/sdl_common_internal.h:22,
  cargo:warning=                 from /home/tnewsome/projects/tmp/lvgl-rs/lvgl-sys/vendor/lv_drivers/sdl/sdl_common.c:8:
  cargo:warning=/usr/include/SDL2/SDL_config.h:4:10: fatal error: SDL2/_real_SDL_config.h: No such file or directory
  cargo:warning=    4 | #include <SDL2/_real_SDL_config.h>
  cargo:warning=      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
  cargo:warning=compilation terminated.
  exit status: 1
```

I see that you already added an `apt get install` line for those headers to Cross.toml, but for whatever reason I'm not getting them. Maybe I do need a newer docker. (I upgraded from Ubuntu LTS to Ubuntu latest release, but that did not bring a significantly newer docker.)

nia-e, 2023-03-25:
This is caused by latest PR merge #64 which still has some WIP areas. Set the environment variable `LVGL_LINK=SDL2` and try again, or disable the `drivers` feature.

rtwfroody, 2023-03-26:
Thanks. I managed to build an example with:
```
CROSS_CONTAINER_ENGINE_NO_BUILDKIT=1 \
    DEP_LV_CONFIG_PATH=`pwd`/examples/include \
    LVGL_LINK=SDL2 \
    cross build \
        --target=armv7-unknown-linux-gnueabihf --example demo --features="alloc" \
        #--release
```

Running that binary gets me a segfault, which could be due to a mismatch in the SDL binary that was linked against. This is probably not relevant for this ticket. The first valgrind error is
```
==27024== Memcheck, a memory error detector
==27024== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==27024== Using Valgrind-3.14.0 and LibVEX; rerun with -h for copyright info
==27024== Command: ./demo
==27024== 
Before all widgets: lv_mem_monitor_t { total_size: 1048576, free_cnt: 2, free_size: 1043260, free_biggest_size: 1043236, used_cnt: 72, max_used: 1932, used_pct: 1, frag_pct: 1 }
==27024== Invalid read of size 1
==27024==    at 0x13DE12: lv_disp_get_hor_res (lv_hal_disp.c:352)
==27024==    by 0x14777F: _lv_inv_area (lv_refr.c:225)
==27024==    by 0x143153: lv_obj_invalidate_area (lv_obj_pos.c:848)
==27024==    by 0x143207: lv_obj_invalidate (lv_obj_pos.c:864)
==27024==    by 0x144623: lv_obj_refresh_style (lv_obj_style.c:173)
==27024==    by 0x1443AB: lv_obj_add_style (lv_obj_style.c:103)
==27024==    by 0x12215F: lvgl::lv_core::obj::Widget::add_style (obj.rs:41)
==27024==    by 0x11C647: demo::main (demo.rs:49)
==27024==    by 0x11B8A7: core::ops::function::FnOnce::call_once (function.rs:250)
==27024==    by 0x11D3BB: std::sys_common::backtrace::__rust_begin_short_backtrace (backtrace.rs:121)
==27024==    by 0x11B1EF: std::rt::lang_start::{{closure}} (rt.rs:166)
==27024==    by 0x4195A7: call_once<(), (dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe)> (function.rs:287)
==27024==    by 0x4195A7: do_call<&(dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe), i32> (panicking.rs:483)
==27024==    by 0x4195A7: try<i32, &(dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe)> (panicking.rs:447)
==27024==    by 0x4195A7: catch_unwind<&(dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe), i32> (panic.rs:140)
==27024==    by 0x4195A7: {closure#2} (rt.rs:148)
==27024==    by 0x4195A7: do_call<std::rt::lang_start_internal::{closure_env#2}, isize> (panicking.rs:483)
==27024==    by 0x4195A7: try<isize, std::rt::lang_start_internal::{closure_env#2}> (panicking.rs:447)
==27024==    by 0x4195A7: catch_unwind<std::rt::lang_start_internal::{closure_env#2}, isize> (panic.rs:140)
==27024==    by 0x4195A7: std::rt::lang_start_internal (rt.rs:148)
==27024==  Address 0xbd9c9280 is on thread 1's stack
==27024==  57712 bytes below stack pointer
```

nia-e, 2023-03-26:
Doubt this is an SDL issue, that's only used in the `sdl` example. There might be issues on running related to the `embedded-graphics` crate, but they shouldn't be this early. The exact location of this segfault is very odd though, I'll investigate this further and get back to you ASAP

nia-e, 2023-03-27:
After testing a bunch of things, I think you were right about SDL2 being the issue. Running as
```
DEP_LV_CONFIG_PATH=`pwd`/examples/include \
cross run --target=armv7-unknown-linux-gnueabihf --example demo \
--features="alloc embedded_graphics" --no-default-features
```
seems to have it error out with
```
[Info]  (0.000, +0)      lv_init: begin         (in lv_obj.c line #102)
[Info]  (0.000, +0)      lv_obj_create: begin   (in lv_obj.c line #206)
[Info]  (0.000, +0)      lv_obj_create: begin   (in lv_obj.c line #206)
[Info]  (0.000, +0)      lv_obj_create: begin   (in lv_obj.c line #206)
Before all widgets: lv_mem_monitor_t { total_size: 1048576, free_cnt: 2, free_size: 1043260, free_biggest_size: 1043236, used_cnt: 72, max_used: 1932, used_pct: 1, frag_pct: 1 }
[Info]  (0.000, +0)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (0.000, +0)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (0.000, +0)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (0.000, +0)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (0.000, +0)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (0.000, +0)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (0.000, +0)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (0.000, +0)      lv_label_create: begin         (in lv_label.c line #75)
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "No available video device"', /home/nia/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-graphics-simulator-0.4.0/src/window.rs:233:51
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
which I assume is a result of the `cross` environment not being able to access my video out - but that is almost certainly not due to the executable being faulty, but just because it's running in an emulator. Does this fix it for you?

nia-e, 2023-03-28:
This is similar to an issue I was encountering while debugging a few days ago, and in retrospect I think you were bumping up against the same issue. Update to latest master, #74 fixed a very similar issue and #75 fixed something related.

rtwfroody, 2023-03-28:
Success!
[image]

Thanks for all your help.

I noticed that the Cross.toml you submitted does not include
```
[target.armv7-unknown-linux-gnueabihf]
pre-build = [ 
    "ln -fs /usr/share/zoneinfo/Europe/Berlin /etc/localtime",
    "dpkg --add-architecture armhf",
    "apt-get update",
    "DEBIAN_FRONTEND=noninteractive apt-get install --assume-yes tzdata",
    "dpkg-reconfigure --frontend noninteractive tzdata",
    "apt-get --assume-yes install clang libclang",
    "apt-get --assume-yes install libsdl2-dev:armhf"
]
```

Are you planning to add that?

closed 2023-03-28 by nia-e