lvgl/lv_binding_rust #58: Unresolved import error for alloc::sync in cstr_core on target thumbv6m-none-eabi

thecodechemist99, 2023-02-13:
I tried to compile a project with `lvgl = "0.5.2"` and get the following cargo build error:
```
   Compiling cstr_core v0.2.6
error[E0432]: unresolved import `alloc::sync`
  --> /Users/florianbeck/.cargo/registry/src/github.com-1ecc6299db9ec823/cstr_core-0.2.6/src/lib.rs:25:12
   |
25 | use alloc::sync::Arc;
   |            ^^^^ could not find `sync` in `alloc`

For more information about this error, try `rustc --explain E0432`.
error: could not compile `cstr_core` due to previous error
```

As outlined in the [rust docs](https://doc.rust-lang.org/alloc/sync/index.html) the Module `alloc::sync` is not implemented by all platforms and the `thumbv6m-none-eabi` target seems to be one of them.
Is there a way around this?

thecodechemist99, 2023-02-13:
The `cstr_core` [docs](https://docs.rs/crate/cstr_core/0.2.6) mention this as well:

> Some hardware targets (e.g. thumbv6m-none-eabi for Cortex M0,M0+) have no support for atomic operations. For these platforms, disable the arc feature to omit the parts of the crate that depend on atomic operations. Compatibility with thead-safe code and Arc<T> will not be available.

I tried changing the dependency in `lvgl-rs/lvgl/Cargo.toml` to `cstr_core = { version = "0.2.6", no-standard-features = true, features = ["alloc"] }`. Unfortunately this didn't resolve the issue.

thecodechemist99, 2023-02-13:
Well, I'm stupid … I used the wrong flag at first (`no-default-features = true`) and then while trying all sorts of stuff missed the fact that changing it to the right flag actually did solve the issue and just created another one.

With the import `cstr_core = { version = "0.2.3", default-features = false, features = ["alloc"] }` in the `lvgl/Cargo.toml` the crate `cstr_core` now compiles fine.
But instead I now get an error from the custom build command:
```
error: failed to run custom build command for `lvgl-sys v0.5.2 (/Users/thecodechemist99/Repos/pi-pico-mp/lib/lvgl-rs/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/Users/thecodechemist99/Repos/pi-pico-mp/target/debug/build/lvgl-sys-44db868db3eefdf6/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=/Users/thecodechemist99/Repos/pi-pico-mp/src/ui/lv_conf.h

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', lib/lvgl-rs/lvgl-sys/build.rs:136:39
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

The associated stack backtrace:
```
stack backtrace:
     0: rust_begin_unwind
               at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/std/src/panicking.rs:575:5
     1: core::panicking::panic_fmt
               at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/panicking.rs:64:14
     2: core::result::unwrap_failed
               at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/result.rs:1791:5
     3: core::result::Result<T,E>::unwrap
               at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/result.rs:1113:23
     4: build_script_build::add_c_files
               at ./build.rs:136:14
     5: build_script_build::main
               at ./build.rs:61:5
     6: core::ops::function::FnOnce::call_once
               at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/ops/function.rs:507:5
```

rafaelcaricio, 2023-02-14:
Hi @thecodechemist99! Thank you for the well-written issue, that is very helpful. Would you like to open a PR for this?

thecodechemist99, 2023-02-15:
Do you have any ideas regarding the `string.h` error?
I'm not very well versed in C, unfortunately and the bug seems to come from the vendor C code(?).

rafaelcaricio, 2023-02-15:
@thecodechemist99 The error says, "fatal error: 'string.h' file not found". I think it is something with your compilation path. It just needs the C header file "string.h" which is usually found in your OS wherever there is a compiler available.

On a quick search, I could find this header file in a few places in my computer:
```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/string.h
/opt/homebrew/opt/llvm/include/c++/v1/string.h
```

FYI, I'm not actively maintaining this project anymore: https://github.com/rafaelcaricio/lvgl-rs#-lvgl---open-source-embedded-gui-library-in-rust

rafaelcaricio, 2023-02-15:
In case it helps, the implementation of this header is done in Rust and is available here: https://github.com/rafaelcaricio/lvgl-rs/blob/master/lvgl-sys/src/string_impl.rs The linker will use this implementation.

nia-e, 2023-03-28:
Hi! Is this issue still happening? A lot has been churned in this library including an update to LVGL 8.x, so if you're still interested @thecodechemist99 let me know if this is still reproducible. Note that there have been major API-breaking changes since this issue was opened.

nia-e, 2023-03-29:
That's caused by the `drivers` feature. Try running the `demo` example with `--features="alloc embedded_graphics" --no-default-features`, or set the `LVGL_INCLUDE` environment variable to wherever SDL headers are (by default, it should look in `/usr/include` and `/usr/local/include`)

thecodechemist99, 2023-03-29:
Thanks, the demo ran successfully with this:
```
DEP_LV_CONFIG_PATH=`pwd`/examples/include cargo run --example demo --features="alloc embedded_graphics" --no-default-features
```

I changed the settings in my project accordingly to:
```
[features]
# Workaround for lvgl-rs with no_std
default = ["build_dep", "embedded_graphics"]

[env]
DEP_LV_CONFIG_PATH = { value = "src/ui", relative = true }
LVGL_INCLUDE = "/opt/homebrew/include"
```
This resolved the previous error (the homebrew location is different on Apple-Silicon Macs), but the compiler complains about more. Shouldn't this have default features disabled and therefore ignore the `drivers` feature? Doesn’t look like it does:
```
The following warnings were emitted during compilation:
warning: /Users/thecodechemist99/.cargo/git/checkouts/lvgl-rs-27f20f23afade818/c0263a2/lvgl-sys/vendor/lv_drivers/display/fbdev.c:26:10: fatal error: 'linux/fb.h' file not found
warning: #include <linux/fb.h>
warning:          ^~~~~~~~~~~~
warning: 1 error generated.

error: failed to run custom build command for `lvgl-sys v0.5.2 (https://github.com/rafaelcaricio/lvgl-rs?branch=master#c0263a29)`

[…]

warning: build failed, waiting for other jobs to finish...
The following warnings were emitted during compilation:

warning: In file included from /opt/homebrew/include/SDL2/SDL_main.h:25,
warning:                  from /opt/homebrew/include/SDL2/SDL.h:32,
warning:                  from /Users/thecodechemist99/.cargo/git/checkouts/lvgl-rs-27f20f23afade818/c0263a2/lvgl-sys/vendor/lv_drivers/sdl/sdl_common_internal.h:21,
warning:                  from /Users/thecodechemist99/.cargo/git/checkouts/lvgl-rs-27f20f23afade818/c0263a2/lvgl-sys/vendor/lv_drivers/sdl/sdl_common.c:8:
warning: /opt/homebrew/include/SDL2/SDL_stdinc.h:382:42: error: static assertion failed: "sizeof(SDL_DUMMY_ENUM) == sizeof(int)"
warning:   382 | #define SDL_COMPILE_TIME_ASSERT(name, x) _Static_assert(x, #x)
warning:       |                                          ^~~~~~~~~~~~~~
warning: /opt/homebrew/include/SDL2/SDL_stdinc.h:420:1: note: in expansion of macro 'SDL_COMPILE_TIME_ASSERT'
warning:   420 | SDL_COMPILE_TIME_ASSERT(enum, sizeof(SDL_DUMMY_ENUM) == sizeof(int));
warning:       | ^~~~~~~~~~~~~~~~~~~~~~~

error: failed to run custom build command for `lvgl-sys v0.5.2 (https://github.com/rafaelcaricio/lvgl-rs?branch=master#c0263a29)`
```

nia-e, 2023-03-29:
Oof, that's on me. It really should have, I'll make sure to fix that asap. In the meantime try setting line 11 in the driver config to 0 and let me know if it builds.

nia-e, 2023-03-29:
Can you post  your `Cargo.toml`?

nia-e, 2023-03-29:
Maybe the location that was suggested [here](https://github.com/rafaelcaricio/lvgl-rs/issues/58#issuecomment-1432034734) could be used. Enable the `drivers` feature, disable the driver config file as [here](https://github.com/rafaelcaricio/lvgl-rs/issues/58#issuecomment-1489072635), and set `LVGL_INCLUDE = "/opt/homebrew/include,/opt/homebrew/opt/llvm/include/c++/v1"

thecodechemist99, 2023-03-29:
Unfortunately, that yields exactly the same result.

nia-e, 2023-03-29:
Is there a `string.h` file at that location on your device? The README says to install various C libraries, but I'm unsure how they translate to macOS

thecodechemist99, 2023-03-29:
> Is there a `string.h` file at that location on your device?

Yes, there is. Sorry, I forgot to mention that.

nia-e, 2023-03-29:
> Yes, there is. Sorry, I forgot to mention that.
That's... very weird. I'll try a patch on a new branch in a second.
> Edit: Which README?
[Here](https://github.com/rafaelcaricio/lvgl-rs/#system-build-dependencies)

nia-e, 2023-03-29:
Try branch `fix-str-import`

thecodechemist99, 2023-03-30:
Hi,
unfortunately, this doesn’t help either. Still the same error.

Here’s the content of the `string.h` file, in case that helps:
```
// -*- C++ -*-
//===----------------------------------------------------------------------===//
//
// Part of the LLVM Project, under the Apache License v2.0 with LLVM Exceptions.
// See https://llvm.org/LICENSE.txt for license information.
// SPDX-License-Identifier: Apache-2.0 WITH LLVM-exception
//
//===----------------------------------------------------------------------===//

#ifndef _LIBCPP_STRING_H
#define _LIBCPP_STRING_H

/*
    string.h synopsis

Macros:

    NULL

Types:

    size_t

void* memcpy(void* restrict s1, const void* restrict s2, size_t n);
void* memmove(void* s1, const void* s2, size_t n);
char* strcpy (char* restrict s1, const char* restrict s2);
char* strncpy(char* restrict s1, const char* restrict s2, size_t n);
char* strcat (char* restrict s1, const char* restrict s2);
char* strncat(char* restrict s1, const char* restrict s2, size_t n);
int memcmp(const void* s1, const void* s2, size_t n);
int strcmp (const char* s1, const char* s2);
int strncmp(const char* s1, const char* s2, size_t n);
int strcoll(const char* s1, const char* s2);
size_t strxfrm(char* restrict s1, const char* restrict s2, size_t n);
const void* memchr(const void* s, int c, size_t n);
      void* memchr(      void* s, int c, size_t n);
const char* strchr(const char* s, int c);
      char* strchr(      char* s, int c);
size_t strcspn(const char* s1, const char* s2);
const char* strpbrk(const char* s1, const char* s2);
      char* strpbrk(      char* s1, const char* s2);
const char* strrchr(const char* s, int c);
      char* strrchr(      char* s, int c);
size_t strspn(const char* s1, const char* s2);
const char* strstr(const char* s1, const char* s2);
      char* strstr(      char* s1, const char* s2);
char* strtok(char* restrict s1, const char* restrict s2);
void* memset(void* s, int c, size_t n);
char* strerror(int errnum);
size_t strlen(const char* s);

*/

#include <__config>

#if !defined(_LIBCPP_HAS_NO_PRAGMA_SYSTEM_HEADER)
#  pragma GCC system_header
#endif

#include_next <string.h>

// MSVCRT, GNU libc and its derivates may already have the correct prototype in
// <string.h>. This macro can be defined by users if their C library provides
// the right signature.
#if defined(__CORRECT_ISO_CPP_STRING_H_PROTO) || defined(_LIBCPP_MSVCRT) || \
    defined(__sun__) || defined(_STRING_H_CPLUSPLUS_98_CONFORMANCE_)
#define _LIBCPP_STRING_H_HAS_CONST_OVERLOADS
#endif

#if defined(__cplusplus) && !defined(_LIBCPP_STRING_H_HAS_CONST_OVERLOADS) && defined(_LIBCPP_PREFERRED_OVERLOAD)
extern "C++" {
inline _LIBCPP_INLINE_VISIBILITY
char* __libcpp_strchr(const char* __s, int __c) {return (char*)strchr(__s, __c);}
inline _LIBCPP_INLINE_VISIBILITY _LIBCPP_PREFERRED_OVERLOAD
const char* strchr(const char* __s, int __c) {return __libcpp_strchr(__s, __c);}
inline _LIBCPP_INLINE_VISIBILITY _LIBCPP_PREFERRED_OVERLOAD
      char* strchr(      char* __s, int __c) {return __libcpp_strchr(__s, __c);}

inline _LIBCPP_INLINE_VISIBILITY
char* __libcpp_strpbrk(const char* __s1, const char* __s2) {return (char*)strpbrk(__s1, __s2);}
inline _LIBCPP_INLINE_VISIBILITY _LIBCPP_PREFERRED_OVERLOAD
const char* strpbrk(const char* __s1, const char* __s2) {return __libcpp_strpbrk(__s1, __s2);}
inline _LIBCPP_INLINE_VISIBILITY _LIBCPP_PREFERRED_OVERLOAD
      char* strpbrk(      char* __s1, const char* __s2) {return __libcpp_strpbrk(__s1, __s2);}

inline _LIBCPP_INLINE_VISIBILITY
char* __libcpp_strrchr(const char* __s, int __c) {return (char*)strrchr(__s, __c);}
inline _LIBCPP_INLINE_VISIBILITY _LIBCPP_PREFERRED_OVERLOAD
const char* strrchr(const char* __s, int __c) {return __libcpp_strrchr(__s, __c);}
inline _LIBCPP_INLINE_VISIBILITY _LIBCPP_PREFERRED_OVERLOAD
      char* strrchr(      char* __s, int __c) {return __libcpp_strrchr(__s, __c);}

inline _LIBCPP_INLINE_VISIBILITY
void* __libcpp_memchr(const void* __s, int __c, size_t __n) {return (void*)memchr(__s, __c, __n);}
inline _LIBCPP_INLINE_VISIBILITY _LIBCPP_PREFERRED_OVERLOAD
const void* memchr(const void* __s, int __c, size_t __n) {return __libcpp_memchr(__s, __c, __n);}
inline _LIBCPP_INLINE_VISIBILITY _LIBCPP_PREFERRED_OVERLOAD
      void* memchr(      void* __s, int __c, size_t __n) {return __libcpp_memchr(__s, __c, __n);}

inline _LIBCPP_INLINE_VISIBILITY
char* __libcpp_strstr(const char* __s1, const char* __s2) {return (char*)strstr(__s1, __s2);}
inline _LIBCPP_INLINE_VISIBILITY _LIBCPP_PREFERRED_OVERLOAD
const char* strstr(const char* __s1, const char* __s2) {return __libcpp_strstr(__s1, __s2);}
inline _LIBCPP_INLINE_VISIBILITY _LIBCPP_PREFERRED_OVERLOAD
      char* strstr(      char* __s1, const char* __s2) {return __libcpp_strstr(__s1, __s2);}
}
#endif

#endif // _LIBCPP_STRING_H
```

nia-e, 2023-03-30:
Unfortunately I can't look at this more until tomorrow, but off the top of my head, #53 successfully cross-compiled to a similar triple by using `cross`; if nothing there proves useful, I'll spin up a macOS VM and try to debug this.