lvgl/lv_binding_rust #65: Tracking issue for version bump

nia-e, 2023-03-02:
What are the outstanding things to do before bumping to 0.6.0? #51 will be an API-breaking change and so a semver bump will be needed. What else should be added/fixed/reworked?
My ideas thus far:

- [ ] Close #32 
- [ ] Close #42 
- [ ] Merge #51 
- [ ] Merge #64

rafaelcaricio, 2023-03-02:
@nia-e you might find interesting things to upstream from here https://github.com/nviennot/lvgl-rs/commits/master
Usage https://github.com/nviennot/turbo-resin/blob/main/src/ui/main_loop.rs

nia-e, 2023-03-02:
Lovely, thanks! I'll try to work this into the next release then, especially the LVGL update. Might delay #64 more for this.

nia-e, 2023-03-29:
Is there anything else worth doing before we publish 0.6.0? I took a look at #25 but it doesn't seem exceedingly urgent (might make the API a little neater I suppose, so I could do that). For #32, I don't have access to a Windows machine ~~and I'd rather not install malware on my laptop~~ so I'm unsure what I can do. #68 I'm not sure will even be too relevant anymore, between removing the need to call `init()` and not having to deal with the timer once #25 is done.

nia-e, 2023-03-29:
I could write more docs, I suppose - especially on `Widget`s, since we only really have examples there right now.

nia-e, 2023-04-01:
@rafaelcaricio can I bump to 0.6.0 and release? Or are there more things you'd like to see done first? A lot has been updated and other than #82 I don't foresee any real breaking changes happening soon (and that's still a ways off and will require a lot of discussion to get to a final, mergeable state)

rafaelcaricio, 2023-04-02:
@nia-e go for it! ;)

closed 2023-04-02 by nia-e