lvgl/lv_binding_rust #69: UB in safe API if LVGL is not initialized

nia-e, 2023-03-06:
Attempting to call any function that requires memory allocation on the LVGL side before `lvgl::init()` is called will cause undefined behaviour (usually resulting in a segfault). This technically ought to make all memory-allocating functions `unsafe` as safe code can currently call them and trigger this bug. Possibilities I see for fixing this:

### Dynamically check if LVGL is initialized at runtime
Making all LVGL allocating/reading functions call `lvgl_sys::lv_is_initialized()` and error on false is probably the most obvious (and safe) way to fix this. This would definitely prevent any UB and force the caller to actually initialize LVGL instead of giving a cryptic error. However, this would lead to a (small) runtime performance penalty as it requires calling a non-inlineable external function and also we need to remember to do this on all new APIs in the future. We could probably get around the first issue by simply checking the existing `RunOnce` in the initializer, but the 2nd issue remains.
This could also feasibly be done by patching LVGL to check this in its allocator, but I fail to see how that is anything other than an all-around worse form of the above.

### Create a dummy `LvMemHandler`
Moving all memory allocating APIs out of being globally callable and instead making them be methods on a dummy `LvMemHandler` struct whose constructor checks the status of LVGL and initializes it if needed is *probably* the way to go, though this would make the API somewhat less intuitive and harkens back to the `UI` struct. That said, it's probably the option which would lead to the fewest headaches long-term (if only for us).

### Catching `SIGSEGV` and panicking with a more understandable error
This is probably the worst option of the 3 but I'm including it since it has *some* merit - namely, it requires no API compliexification and will lead to no overhead at runtime. However, triggering a segfault means that [UB has already occured](https://en.cppreference.com/w/cpp/language/ub) and therefore exiting the program would just lead to *more* UB by e.g. attempting to drop malformed values or instantiating uninitialized values (if only for a fraction of a second), [which is still UB](https://doc.rust-lang.org/stable/std/mem/union.MaybeUninit.html). This is especially problematic if the resulting binary has elevated privileges or is running on baremetal, since it might mess up the entire system. It would also not let us know if the segfault is a result of user error for not initializing LVGL or some other bug in `lvgl-rs` (or in LVGL itself).

I haven't quite decided what I want to do yet - probably option 2, but I could be convinced option 1 is better. Also, I'd be happy to hear any other suggestions.

cydergoth, 2023-03-07:
This sounds like a classic case for type builder pattern?

nia-e, 2023-03-07:
What specifically are your suggesting?

rafaelcaricio, 2023-03-07:
I've thought A LOT about this in the past. It is one of the major frustrations I had while trying to think about a nice API for this crate.

In the end, I came to peace with the simplest solution possible, which is letting the people using this library remember to call `lvgl::init()` and letting them take the responsibility. Unfortunately, the LVGL C API makes heavy use of global state, and my first attempt with the `lvgl::UI` struct was to try to contain misuse. But that landed us in a very cumbersome API. Since LVGL is meant to be used in constrained environments, adding any overhead at runtime could be a dealbreaker.

cydergoth, 2023-03-07:
So type builder is an extension of the builder pattern in which you return
a different type after various steps of building. In this case, the initial
type would only expose the init methods, but calling that would return a
type with the rest of the methods exposed. This article goes into some more
detail https://dev.to/mindflavor/rust-builder-pattern-with-types-3chf

nia-e, 2023-03-07:
> I've thought A LOT about this in the past. It is one of the major frustrations I had while trying to think about a nice API for this crate.
>
> In the end, I came to peace with the simplest solution [...]

There is another option a friend suggested in private. The `[static_init](https://docs.rs/static_init/latest/static_init/)` crate could allow us to initialize LVGL asap by setting a function which initializes it as something to be executed before `main()` in all crates that compile us in. Since that might be undesirable and we might want to at least *allow* users to not initialize LVGL, we could also e.g. set a `static mut UNSAFE_LVGL_NO_AUTO_INIT: bool = false` with priority 0 in `static_init` and then call `lvgl::init()` if it's not true with priority 2 (thus allowing the user to unsafely set it to `true` with priority 1). We could perhaps provide a macro for this (again, clearly marking as unsafe).

nia-e, 2023-03-07:
> So type builder is an extension of the builder pattern in which you return a different type after various steps of building. In this case, the initial type would only expose the init methods, but [...]

Ah, thanks! But I worry this gets into the same issue as the `UI` struct we used to have, namely that the API ends up being grossly unintuitive and a far cry from the neat LVGL API

PR #79 (2023-03-29): Auto-initialize LVGL
As discussed in #69, the current API can cause UB if not used properly. The option of using `static_init` to auto-initialize was agreed on, though I opted for [`ctor`](https://crates.io/crates/ctor) since it does the same thing but seems better-maintained. I added a feature also to gate disabling this behaviour, along with a huge disclaimer as to why it shouldn't be used.

Closes #69