lvgl/lv_binding_rust #100: Tracking issue for `lvgl-codegen` and general code generation rewrite

nia-e, 2023-04-06:
This is a *huge* project and would lead to probably simplifying our lives a lot, but also will take a lot of brainpower to finish. My ideas would be to create *something* that allows for automatic code generation at compile-time so as to not require us to manually implement functions on e.g. widgets (and maybe more later). This would also probably harden us significantly against LVGL version bumps.

This is likely the most ambitious idea I've submitted yet. I plan to get started on this once everything else for 0.7 is done, or thereabouts (see #97). I will add relevant PRs and issues as they appear.

rafaelcaricio, 2023-04-07:
If we have a well-defined way of how widgets would look like in Rust then we can potentially write down the rules of conversion. This is what I was trying to do when writing the first version of the codegen. The greater problem, for me, was trying to do both (define common patterns and write down the codegen) at the same time. 😬

nia-e, 2023-04-07:
I did notice there were some stragglers in the code (the `analysis.rs` module, for instance) - that's where I got the inspiration in fact. I might end up writing something in Coq and use that to sketch out the rules of generation - if so I'll make sure to drop a link to that repository here.