lvgl/lv_binding_rust #103: Can this cargo work with gtk ?

vortex314, 2023-04-30:
Changing the lv_drv_cmore lv_cronf.h to work with GTK makes that it doesn't compile anymore.
Is this intentional ?
I have libgtk-3.0 dev installed.

My intention is to write and test a GUI dashboard under Ubuntu and then run it on raspberry pi using the framebuffer.
```
  running: "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-gdwarf-4" "-fno-omit-frame-pointer" "-m64" "-I" "/home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/lvgl-sys-0.6.2/vendor/lvgl/src" "-I" "/home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/lvgl-sys-0.6.2/vendor" "-I" "/home/lieven/workspace/lv_1" "-I" "/home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/lvgl-sys-0.6.2/vendor/lv_drivers" "-I" "/usr/include" "-I" "/usr/local/include" "-DLV_CONF_INCLUDE_SIMPLE=1" "-o" "/home/lieven/workspace/lv_1/target/debug/build/lvgl-sys-5f991eed6eeaed4e/out/20eb48ef7c011711-gtkdrv.o" "-c" "/home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/lvgl-sys-0.6.2/vendor/lv_drivers/gtkdrv/gtkdrv.c"
  cargo:warning=/home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/lvgl-sys-0.6.2/vendor/lv_drivers/gtkdrv/gtkdrv.c:12: warning: "_DEFAULT_SOURCE" redefined
  cargo:warning=   12 | #define _DEFAULT_SOURCE /* needed for usleep() */
  cargo:warning=      | 
  cargo:warning=In file included from /usr/include/x86_64-linux-gnu/bits/libc-header-start.h:33,
  cargo:warning=                 from /usr/include/stdint.h:26,
  cargo:warning=                 from /usr/lib/gcc/x86_64-linux-gnu/11/include/stdint.h:9,
  cargo:warning=                 from /home/lieven/workspace/lv_1/lv_conf.h:20,
  cargo:warning=                 from /home/lieven/workspace/lv_1/lv_drv_conf.h:16,
  cargo:warning=                 from /home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/lvgl-sys-0.6.2/vendor/lv_drivers/gtkdrv/gtkdrv.h:18,
  cargo:warning=                 from /home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/lvgl-sys-0.6.2/vendor/lv_drivers/gtkdrv/gtkdrv.c:9:
  cargo:warning=/usr/include/features.h:236: note: this is the location of the previous definition
  cargo:warning=  236 | # define _DEFAULT_SOURCE        1
  cargo:warning=      | 
  cargo:warning=/home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/lvgl-sys-0.6.2/vendor/lv_drivers/gtkdrv/gtkdrv.c:15:10: fatal error: gtk/gtk.h: No such file or directory
  cargo:warning=   15 | #include <gtk/gtk.h>
  cargo:warning=      |          ^~~~~~~~~~~
  cargo:warning=compilation terminated.
  exit status: 1

```

nia-e, 2023-04-30:
Hi!
GTK support is currently untested; only SDL2 is fully confirmed to be working for the drivers and should be considered "ready". However, I'm unsure why this specific issue is occuring; setting the `LVGL_INCLUDE` environment variable manually may remedy this issue though (by default, it will be set to `/usr/include,/usr/local/include`). See [the docs](https://docs.rs/lvgl/latest/lvgl/drivers/index.html) for details. Also, I had experienced some issues with building the GTK bindings in the past and I suspect there may be issues upstream. Getting this fixed will be added to the roadmap for 0.7 though so thank you for the logs!

vortex314, 2023-04-30:
Yeah, it looks I am one of the first to try.
Compiling succeeds with 
'''
export DEP_LV_CONFIG_PATH=`pwd`
export LVGL_INCLUDE=/usr/include/gtk-3.0,/usr/include,/usr/local/include,/usr/include/glib-2.0
export LVGL_INCLUDE=$LVGL_INCLUDE,/usr/lib/x86_64-linux-gnu/glib-2.0/include,/usr/include/pango-1.0
export LVGL_INCLUDE=$LVGL_INCLUDE,/usr/include/harfbuzz,/usr/include/cairo,/usr/include/gdk-pixbuf-2.0,/usr/include/atk-1.0,/usr/include/freetype2
export LVGL_LINK=gtk-3,gdk-3,pangocairo-1.0,pango-1.0,harfbuzz,atk-1.0,cairo-gobject,cairo,gdk_pixbuf-2.0,gio-2.0,gobject-2.0,glib-2.0
export LIBCLANG_PATH=$(llvm-config --libdir)
cargo run
'''

cydergoth, 2023-04-30:
Thanks for working together on this as it is one of the scenarios I am also
interested in, especially under OpenWRT (x86)

vortex314, 2023-04-30:
Now stuck at setting the color_depth correct . In GTK linux this is 32 bit, bu the crate doesnùt adapt when chaning this in lv_conf.h
```
/*Color depth: 1 (1 byte per pixel), 8 (RGB332), 16 (RGB565), 32 (ARGB8888)*/
#define LV_COLOR_DEPTH 32
```
Compiler error
```
   Compiling lvgl-sys v0.6.2
   Compiling lvgl v0.6.2
error[E0308]: mismatched types
   --> /home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/lvgl-0.6.2/src/display.rs:262:29
    |
262 |         disp_drv.flush_cb = flush_cb;
    |                             ^^^^^^^^ expected `lv_color32_t`, found `lv_color16_t`
    |
    = note: expected enum `Option<unsafe extern "C" fn(_, _, *mut lv_color32_t)>`
               found enum `Option<unsafe extern "C" fn(_, _, *mut lv_color16_t)>`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `lvgl` due to previous error
```

nia-e, 2023-04-30:
> Yeah, it looks I am one of the first to try. Compiling succeeds with
>
> ```
> export DEP_LV_CONFIG_PATH=`pwd`
> export [...]

Thanks! This should be configured automatically once #82 is merged.

> Now stuck at setting the color_depth correct . In GTK linux this is 32 bit, bu the crate doesnùt adapt when chaning this in lv_conf.h
>
> ```
> /*Color depth: 1 (1 byte per pixel), 8 (RGB332), 16 [...]

I think I know the issue here. I'll merge a fix tomorrow and give you a ping.

> Additionl challenge, my sample creates the buffer on stack, so I got a LInux stack overflow with this code
> 
> ```
>     const HOR_RES: u32 = 1024;
>     const VER_RES: u32 = 768;
> 
>     let buffer = DrawBuffer::<{ (HOR_RES * VER_RES) as usize }>::default();
> ```

This is a known issue with creating large `DrawBuffer`s, I'm unsure how to approach fixing this. However, you can make the buffer smaller; the LVGL docs [claim](https://docs.lvgl.io/8.3/porting/display.html) that it should be identical performance-wise down to the buffer being 1/10th the size of the display.

vortex314, 2023-04-30:
@nia-e  great ! Thanks for the swift feedback. Looking forward to the evolution.

vortex314, 2023-04-30:
For what it's worth , got so far 
[image]

nia-e, 2023-04-30:
Guessing that's the result of the horribly broken color depth? Still, glad to see GTK (almost) working :D

nia-e, 2023-04-30:
Ah, forgot to add - the bottom text is configurable in`lv_conf.h`. It's on by default in the example file for debugging purposes.

vortex314, 2023-04-30:
Concerning the stack overflow, this can be patched  by the command under linux : 
```
ulimit -s 30000
```
This increase the stack from 8 MB to 30 MB . And permitted to have a resolution of 1024x768. 
A better solution would be to create this buffer on the heap. 
[image]

nia-e, 2023-04-30:
The buffer does live on the heap; the issue is just that on creation, it is (briefly) allocated on the stack. The relevant code is [here](https://github.com/lvgl/lv_binding_rust/blob/master/lvgl/src/display.rs#L153). Feel free to suggest fixes, but the buffer-related code is probably the most radioactive code in this entire project so be warned

nia-e, 2023-05-01:
Hi @vortex314,
A possible fix for the colourspace issue should be on the branch in #104. Can you give this a test and let me know if it runs successfully?

vortex314, 2023-05-01:
Thanks for the fast update. Just for future reference I simplified build by using .cargo/config.toml
```
[env]
DEP_LV_CONFIG_PATH={ value=".",relative=true}
LVGL_INCLUDE = """
/usr/include/gtk-3.0,\
/usr/include,\
/usr/local/include,\
/usr/include/glib-2.0,\
/usr/lib/x86_64-linux-gnu/glib-2.0/include,\
/usr/include/pango-1.0,\
/usr/include/harfbuzz,\
/usr/include/cairo,\
/usr/include/gdk-pixbuf-2.0,\
/usr/include/atk-1.0,\
/usr/include/freetype2"""
LVGL_LINK="gtk-3,gdk-3,pangocairo-1.0,pango-1.0,harfbuzz,atk-1.0,cairo-gobject,cairo,gdk_pixbuf-2.0,gio-2.0,gobject-2.0,glib-2.0"
LIBCLANG_PATH={ value = "/usr/lib/llvm-14/lib" ,force=true}
# the above path can be obtained with >llvm-config --libdir<  , don't know how to do get the ouput of a command in env var
```

vortex314, 2023-05-01:
Looks fine now, Thanks for the help.

closed 2023-05-01 by vortex314

nia-e, 2023-05-08:
Re: the stack overflow mentioned in this issue; it should be fixed as soon as #117 is merged. I think it's ready but I'll wait for tomorrow to merge it just in case anyone spots an issue with it (there have been *many*)