lvgl/lv_binding_rust #105: How to create a Grid layout ?

vortex314, 2023-05-01:
How can I create a Grid Layout ? 
 As this is based on LVGL 8.3 , I'm looking for call lv_obj_set_layout(obj, LV_LAYOUT_GRID)
I got the impression, this is part of the Style 
but which parameter do I give : screen_style.set_layout(??)
Or is it something like : lv_obj_set_grid_dsc_array(obj, col_dsc, row_dsc).

Would be nice to have an example like that.

nia-e, 2023-05-01:
Hi!
You can use the `set_layout()` method on a `Style` object. The `LV_LAYOUT_GRID` static is reexported by `lvgl_sys`, and so the following should work:

```rust
use lvgl::style::Style;

let mut example_style = Style::default();
example_style.set_layout(lvgl_sys::LV_LAYOUT_GRID);
```

and then add the style to an object using the `add_style()` method.

vortex314, 2023-05-01:
The compiler doesn't seem to like it. 
```
error[E0133]: use of mutable static is unsafe and requires unsafe function or block
  --> src/main.rs:58:29
   |
58 |     screen_style.set_layout(LV_LAYOUT_GRID);
   |                             ^^^^^^^^^^^^^^ use of mutable static
   |
   = note: mutable statics can be mutated by multiple threads: aliasing violations or data races will cause undefined behavior
```
Putting this in an unsafe block works.

vortex314, 2023-05-01:
BTW : how do I set the array of the grid ? Github copilot suggest this : 
```
        screen_style.set_layout_grid_col_dsc_array(&[100, 200, 100]);
        screen_style.set_layout_grid_row_dsc_array(&[100, 200, 100]);
```
This doesn't compile. I got the impression that github copilot or chatgpt don't have enough Rust code to their disposal to work with.

nia-e, 2023-05-01:
> The compiler doesn't seem to like it.
>
> ```
> error[E0133]: use of mutable static is unsafe and requires unsafe function or block
>   --> src/main.rs:58:29
>    |
> 58 | [...]

Bah, I'll put a safe wrapper around it.

> BTW : how do I set the array of the grid ? Github copilot suggest this :
>
> ```
>         screen_style.set_layout_grid_col_dsc_array(&[100, 200, 100]); [...]

It had approximately the right idea, but those 2 functions are currently not implemented. I can get them done tomorrow, shouldn't be too complex.

vortex314, 2023-05-01:
Great thanks ! You're better than any AI ! :+1:

nia-e, 2023-05-02:
Hi @vortex314! Can you test out the branch in #106? Should add the necessary APIs on `Style` objects (`set_layout_grid_column_dsc_array()` and `set_layout_grid_row_dsc_array()`), and it also reworks `set_layout()` to be safe (pass it a `Layout::flex()` or `Layout::grid()`)

vortex314, 2023-05-02:
Hi @nia-e  the arguments to set_layout_grid_column_dsc_array and set_layout_grid_row_dsc_array should be an array of sizes of each column and row , no ? Terminated with a magic value. 
The CoorDesc only permit 3 values. 
See : 
```
static lv_coord_t column_dsc[] = {100, 400, LV_GRID_TEMPLATE_LAST};   /*2 columns with 100 and 400 ps width*/
static lv_coord_t row_dsc[] = {100, 100, 100, LV_GRID_TEMPLATE_LAST}; /*3 100 px tall rows*/
```
on https://docs.lvgl.io/8.3/layouts/grid.html#grid-descriptors
Now, it just core dumps, if I try something like this : 
```
    screen_style.set_layout(Layout::grid());
    let x_grid = CoordDesc::from_values(100, 100, 100);
    let y_grid = CoordDesc::from_values(100, 100, 100);
    screen_style.set_grid_column_dsc_array(x_grid);
    screen_style.set_grid_row_dsc_array(y_grid);
```

nia-e, 2023-05-02:
Hi! Can you provide your `lv_conf.h` and `lv_drv_conf.h` files also? I'm going to try to do a deep dive into this but it might take a while.

vortex314, 2023-05-02:
You can find everything here :  https://github.com/vortex314/lv_1.git
Just check the README on some dependencies

nia-e, 2023-05-02:
Alright, thanks. I'll give you a ping when there's a new patch to try.

nia-e, 2023-05-02:
For reference, this is what I see:

[image]

vortex314, 2023-05-02:
Thanks, that seem to be working. Just the result for the same code differs widely. Maybe gtk related.
I run this on Lubuntu 22.04.2 LTS (Jammy Jellyfish), gtk-3.0
[image]

vortex314, 2023-05-02:
Oops, missed that one. Now I got the same result. Thanks for the quick responses !

closed 2023-05-02 by vortex314