lvgl/lv_binding_rust #111: Dropping `Screen`s and associated memory leaks

nia-e, 2023-05-06:
Currently we provide no way to delete a `Screen` after it's been instantiated. Having the user manually call a deletion function also feels... un-rusty and generally unwise as it would be easy to forget. Also, making the screen be deleted on `Drop` carries the risk of e.g. a user calling `get_scr_act()` multiple times and double-deleting a screen -> a double free -> undefined behaviour. I'm honestly kind of stumped as to what to do here; we can't store some kind of `is_active: bool` field either as every call to `get_scr_act()` returns a *different* `Screen` and there's no way to dynamically check on drop if there are other `Screen`s pointing to the same underlying object.

The other option is to have the `Display` store a copy of a pointer to the last `Screen` it returned and simply refuse to return multiple identical `Screen`s thus allowing us to delete on drop, but this does make things slightly more inconvenient.

nia-e, 2023-05-06:
Thinking about it, we can also modify the API so as to require a reference to the `Screen` on widget/object creation (and not get just the active screen) and then store a `PhantomData<&'a T> where T: NativeObject` such that they will automatically be invalidated on drop.

cydergoth, 2023-06-16:
This seems to be the case for any Widget created, unfortunately my understanding of Rust FFI isn't yet good enough to help with this :-(

cydergoth, 2023-06-17:
# Some thoughts on lifecycle management for LVGL resources in rust

## The problem

LVGL is a `C` language libray which uses either the global malloc/free
or a custom allocator. When an LVGL resource is created memory is
allocated on the heap and a constant mutable pointer is returned
(ignoring static styles for the purposes of this discussion).

Some LVGL resources hold pointers to other LVGL resource, e.g. a
widget to the associated set of styles and containers to their
children. In this case, the referenced memory must have a lifecycle
equal to or greater than the containing resource.

When binding to LVGL as a rust FFI library, then the binding needs to
provide appropriate support for the rust user to manage these
resources. This includes creation, lifecycle and then destruction.

Errors to mitigate include:

- Use before allocation. Pointers should be null or valid.
- Dereference null pointer. Null pointers should not be usable in a context
  where they may be dereferenced
- Use after free. A referenced resource must not be freed whilst existing
  references still point to it. References which point to a freed resource
  must be unusable (or null-ed) after free.
- Memory leak. A referenced object should have a defined lifecycle and
  should not exceed that lifecycle
- Multiple concurrent access from different threads

## Usage patterns for resources in LVGL

LVGL has a number of different ways in which resources are used.

### Global lifespan

The most common usage pattern in an _embedded systems_ context is to
allocate LVGL displays, callbacks, screens, styles and widgets early
in the startup sequence of the embedded application and never release
them. This is sufficient for many embedded system UI requirements. In
this scenario, styles may be statically allocated in read-only memory
to save ram. For resources which require dynamic allocation, the lower
level `lvgl_sys::` API may be used.

Where this patten is usable, it does not require free-ing of certain
LVGL resources. The common exception is animations. Pagination, tab or
carousel type requirements may be satisfied by pre-allocated
`Fragments` and the `FragmentManager`.

### Dynamic lifespan

In this usage pattern, which may be required either to low ram
necessitating more fine grained management of ram resource, or in
scenarios where the UI layout and components may not be known at
firmware build time, then LVGL resources including screens may be
dynamically allocated and deallocated during the runtime of the
application.

### Mixed lifespan

This mode is a combination of the Global and Dynamic lifespans in
which some resources exist for the entire lifetime of the application
and some are created and destroyed dynamically.

### Resource ownership

In LVGL, a widget is (with one exception) always parented by another
Widget. Parentage is a one-to-many relationship, and when the parent
widget is deallocated then the children are also deallocated in a
cascade delete pattern.
[... 173 lines cut ...]
  that there will not be rust structs with dangling references to
  memory in the `C` domain. Similar strategies may be required for
  other resources on the rust side which hold FFI pointers.
- Styles must also be implemented using ARC pointers, however they are
  not owned by widgets as they may have a many-to-many relations with
  widgets. This also applies to Fonts, although fonts are usually in
  read-only memory and would have a no-op drop function. To ensure
  that styles exist for the duration of the referencing resources, we
  should either keep a copy of the style references in the Obj, which
  involves handling potentially multiple styles per widget and thus an
  array of Arc references, or in a global table.
- Care must be taken of the need to support `#[no_std]` for embedded
  systems and thus `alloc` needs to be handled internally in the
  bindings libary for that case.
- Other resources may include associated CStrings such as `Label` text.
- These capabilities may be placed behind feature gates to disable
  them in contexts such as Global lifespan.

Side effects of this proposal include:
- Increased memory consumption on the rust side for additional fields in Object
- Increased indirection for ARC pointers
- Additional complexity and memory with ARC+Mutexes, thus requiring
  the embedded HAL to support mutexes.
- This would be a breaking change for existing code using the
  lv_binding_rust library.

nia-e, 2023-06-17:
Hey @cydergoth, thank you so much for the detailed writeup! It definitely echoes some concerns I've had also. I mostly agree that we need to tackle dropping somehow and it's been a big headscratcher for me; I considered something tangentially related in #107 (specifically, [this](https://github.com/lvgl/lv_binding_rust/issues/107#issuecomment-1547783288)) which vaguely correlates.

On `Obj`, I have tried the approach of holding it in a `Box<_>` instead of just the pointer, but we currently don't handle dropping children of dropped objects properly and it resulted in segfaults all over the place. However, I'm not entirely certain that wrapping everything in an `Arc` is a great idea simply for performance reasons; this actually [sort of used to be how we did things](https://github.com/lvgl/lv_binding_rust/commit/f4618fd635cdb3744d9a02be9a2c85fa355e791c) (though only for examples, so it was up to the user to decide how and what to do). Keeping everything behind fat pointers and dynamically handling memory has performance and memory overhead, and LVGL is supposed to run on baremetal with minimal hardware. Reference counting and mutexes are not particularly taxing, but costs add up. Keeping a global mutex is probably less memory-intensive but might degrade actual performance by a lot.

I do like the idea of keeping a global list of live objects, though we are burdened by the fact that - as you pointed out - we don't have an allocator by default except for LVGL's (which has access to very little memory). I'm unsure thus if it's possible to implement it for all cases (i.e. without `alloc`).

I do think there's a happy compromise possible to allow dropping and also not tax resources or performance too much. Rust's lifetimes do allow us to automatically invalidate a lot of things; changing e.g. `Obj` to hold `raw: Box<_>, parent: PhantomData<&'a Obj>` which is recursively invalidated on parent deletion might get us most of the way towards dynamic dropping (and we can simply have users wrap objects they want to live globally as `ManuallyDrop<Obj>`).

This doesn't address threading though, so I am curious what can be done there (current approach being "nothing, just don't touch LVGL-space from multiple threads"). There's also the issue of "what do we do if something manifests an object pointer out of thin air" which can occur (see the animation PR I recently merged, where via witchcraft I need to turn a `*mut lv_obj_t` into a `&mut Obj` where the original `Obj` is then leaked - leaking is useless now, but I specifically did it that way to be resilient against the future possibility of implementing real dropping). Here a global pointer table could be useful, though *maybe* we could also just pretend any witchcraft-derived `Obj` is `'static` or make it `ManuallyDrop` and never drop it since outside of special-case exemptions it can "only" happen if it's secretly a duplicate of an existing object; but then we must somehow ensure said object isn't dropped before this ghost one is, so we need to make sure these ghost objects only exist in their own very short scope.

Honestly I'm not sure what to do, but I worry about the cost involved in dynamic checking of anything. Though if you want to, I would be curious to see benchmarks.

cydergoth, 2023-06-17:
So the performance of rendering *should* be mostly on the `C` side, which would be using all native pointers. The extra overheard of Arc should only come into play when you need to make a call via the rust domain, which should be mostly during UI creation, right? After that it would just be small updates, with LVGL doing all the Rasterizing and rendering. 

The mutex would need to be locked during rendering, then any events would have to wait for rendering to complete, but it would stop one thread like an event thread trying to update the LVGL model whilst another thread is deleting that object 😀 

If you take a look at my repo, you'll see it uses an unsafe impl Send and async to update the UI. I don't think I could do that without async, unless I move completely to a command model and have the render thread poll a command queue for mutation requests - which whilst possible seems a little clunky as async effectively does the poll

nia-e, 2023-06-17:
`Send` was actually implemented for `Obj` until a few days ago when I intentionally removed it cause I realised it could be dangerous; maybe creating a distinct `LvThreaded<T> where T: Widget` and allowing users to work with *those* behind mutexes etc might work?

cydergoth, 2023-06-18:
Do you think prototyping with Arc would be valuable to determine precisely what the tradeoffs are?

nia-e, 2023-06-19:
Sure. Though if there is *any* noticeable impact on memory use or speed, we will probably need this as an explicit opt-in

cydergoth, 2023-06-19:
Yes, it definitely needs a feature gate, as it will need alloc and maybe
mutex too.

nia-e, 2023-06-20:
Would you be interested in working on this with me? I can ask that you be added as a collaborator

cydergoth, 2023-06-20:
I'm a little busy with my actual job rn, but I'm hoping to get a chance to
look at the weekend

nia-e, 2023-06-20:
That's fair. This probably won't be getting merged "soon", but I'll open a dedicated issue for threading and we can discuss it there.

nia-e, 2023-06-20:
See #140. I'll get started on trying to figure things out, but it's probably going to be a while before it works. But in case you want to help out with code - @kisvegabor can you give @cydergoth write perms on branches? I can't manage access for people w/ my current perms.

kisvegabor, 2023-06-21:
@nia-e I've added Admin rights for you. Now you should be able to invite people and change permissions.

nia-e, 2023-06-21:
Thanks! :D