lvgl/lv_binding_rust #118: Is there a way to display GIFs?

bachrc, 2023-05-08:
Hello there, I am searching for quite some time if this is possible to display some gifs, or if i'm doomed to use Micropython?

And if this is not possible, would you know what does those bindings do not bind?

nia-e, 2023-05-08:
Hi! Image embeds aren't supported yet. Take a look at the project's README for a list of supported features.

bachrc, 2023-05-08:
Thanks a lot for your answer. Is there a way to check when this would be implemented, like a roadmap or an issue?

nia-e, 2023-05-08:
It's planned, but it will be a bit until it's implemented as there's some more urgent things to fix right now and I'm the only active maintainer so the rate of work can be a bit erratic. You can keep this issue open to track progress if you'd like, or I can open a dedicated tracking issue once I start work.

bachrc, 2023-05-08:
Thanks a lot! 
For now, I think I'm gonna make a Nix environment with Rust, Python and SDL, and leverage the display to some Python code I will be importing in Rust.

andelf, 2023-06-07:
Check out https://github.com/andelf/tinygif