lvgl/lv_binding_rust #122: Table in Rust misses some methods compared to C code

vortex314, 2023-05-09:
Hi, here again with next one. ;-)  Your swift feedback keeps me motivated to make my application work with LVGL. 
The following code seems to miss the methods ```set_col_width`` and set ```set_row_height``` while they are exposed in the C source. https://docs.lvgl.io/master/widgets/table.html#width-and-height
```
    {
        let mut table = Table::create(&mut cont)?;
        let table_style = new_grid_style(&mut styles, 7, 2, 6, 6);
        table.add_style(Part::Main, table_style)?;
        table.set_col_cnt(3)?;
        table.set_row_cnt(3)?;
        table.set_col_width(0, 100)?;
        table.set_col_width(1, 100)?;
        table.set_col_width(2, 100)?;
        table.set_row_height(0, 50)?;
        table.set_row_height(1, 50)?;
        table.set_row_height(2, 50)?;
        table.set_cell_value(0, 0, & mut CString::new("1").unwrap())?;
        table.set_cell_value(0, 1, &CString::new("2").unwrap())?;
        table.set_cell_value(0, 2, &CString::new("3").unwrap())?;
        table.set_cell_value(1, 0, &CString::new("4").unwrap())?;
        table.set_cell_value(1, 1, &CString::new("5").unwrap())?;
        table.set_cell_value(1, 2, &CString::new("6").unwrap())?;
        table.set_cell_value(2, 0, &CString::new("7").unwrap())?;
        table.set_cell_value(2, 1, &CString::new("8").unwrap())?;
        table.set_cell_value(2, 2, &CString::new("9").unwrap())?;
        table.on_event(|_table, _event| {
            match _event {
                Event::Clicked => {
                    info!("Table clicked");
                }
                _ => {}
            }
        })?;

    }
```
Secondary question : how can I detect which cell was clicked ?  I this via ```lv_table_get_selected_cell``` or via ```lv_event_get_draw_part_dsc``` ? 
Thanks again for your enduring assistance.

nia-e, 2023-05-10:
Hi! Opened a PR to fix this. Presumably, `get_selected_cell` would be the correct function to call. See #123 and let me know if it's fine as usual.

vortex314, 2023-05-10:
Looks you missed :  tabel.set_row_height in the wrapper.

nia-e, 2023-05-10:
There is no height-setting function; in the link you provided (https://docs.lvgl.io/master/widgets/table.html#width-and-height) it states that height is auto-calculated based on other info, and also there is no `lv_table_set_row_height()` function at all.

vortex314, 2023-05-10:
indeed you're right. My bad.

vortex314, 2023-05-10:
In version 7.11 there was a call lv_table_set_cell_type which permitted to later on apply a different style to some cells in the table. I didn't find the equivalent in 8.3 yet. 
The target is that the head row is colored and can be used upon clicking to sort the data.

vortex314, 2023-05-10:
Looks to be decommissioned the, was referring to this : https://docs.lvgl.io/7.11/widgets/table.html?highlight=lv_table_set_cell_type#_CPPv422lv_table_set_cell_typeP8lv_obj_t8uint16_t8uint16_t7uint8_t

nia-e, 2023-05-10:
unsure what I can do about that :c

vortex314, 2023-05-10:
Starts to look good. :-D
[image]

closed 2023-05-10 by vortex314

nia-e, 2023-05-10:
Merged #123

vortex314, 2023-05-10:
This code 
```
    table.on_event(|mut _table, _event| match _event {
        Event::Clicked => {
            let (r, c) = _table.get_selected_cell().unwrap();
            info!("Table event : {:?} row {} col {} ", _event, r, c);
```
Returns for all cells 65535 ? Any cell I can click upon. 
```
2023-05-10 18:52:07.576 INFO lv_1::view - Table event : Clicked row 65535 col 65535 
```

```

Did I miss something ?

nia-e, 2023-05-10:
uhhh... that's odd. I'll take a look

vortex314, 2023-05-10:
Indeed. You're right.
```
2023-05-10 19:07:37.679 INFO lv_1::view - Table event : Clicked row 65535 col 65535 
2023-05-10 19:07:38.436 INFO lv_1::view - Table event : Pressed
2023-05-10 19:07:38.436 INFO lv_1::view - Table event : Pressed row 5 col 1 
2023-05-10 19:07:38.436 INFO lv_1::view - Table event : Pressing
2023-05-10 19:07:38.436 INFO lv_1::view - Table event : Pressing row 5 col 1 
2023-05-10 19:07:38.473 INFO lv_1::view - Table event : Pressing
2023-05-10 19:07:38.474 INFO lv_1::view - Table event : Pressing row 5 col 1 
2023-05-10 19:07:38.504 INFO lv_1::view - Table event : ValueChanged
2023-05-10 19:07:38.504 INFO lv_1::view - Table event : ValueChanged row 5 col 1 
2023-05-10 19:07:38.504 INFO lv_1::view - Table event : Released
2023-05-10 19:07:38.504 INFO lv_1::view - Table event : Released row 65535 col 65535 
```
It's not under the Clicked event

PR #123 (2023-05-10): Flesh out `Table` widget
Adds some more methods needed for using this widget

Closes #122