lvgl/lv_binding_rust #124: What is the Event type for LV_EVENT_DRAW_PART_END  equivalent ?

vortex314, 2023-05-10:
Referring to this call : https://github.com/lvgl/lvgl/blob/630da9c6cac169a294b6be3d2d95a49a4133ce00/examples/widgets/table/lv_example_table_2.c#L84
```
    /*Add an event callback to to apply some custom drawing*/
    lv_obj_add_event_cb(table, draw_event_cb, LV_EVENT_DRAW_PART_END, NULL);
```
As this looks the way to to custom coloring of the table rows. 
The Rust code doesn't seem to receive these kind of events.

nia-e, 2023-05-14:
Implemented in #126, sorry it took so long 😓

closed 2023-05-15 by nia-e

vortex314, 2023-05-16:
Is the below code  the correct way to receive these events ?  As I didn't see the events LV_EVENT_DRAW_PART_END  arriving  when the table is shown. 
``` 
 table.on_event(|mut _table, _event| match _event {
        Event::ValueChanged => {
            let (row, col) = _table.get_selected_cell().unwrap();
            if row == 0 {
                match col {
                    0 => sorting = Sorting::OnTopic,
                    1 => sorting = Sorting::OnValue,
                    2 => sorting = Sorting::OnCount,
                    3 => sorting = Sorting::OnTime,
                    _ => {}
                }
                info!("Sorting on {:?}", sorting);
                send.send(Message::Refresh).unwrap();
            }
        }
        _ => {
            info!("Event {:?}", _event);
        }
    })?;
```
Or is there an alternate way to reach the same as the C code to filter on certain events : 
```
    /*Add an event callback to to apply some custom drawing*/
    lv_obj_add_event(table, draw_part_event_cb, LV_EVENT_DRAW_PART_BEGIN, NULL);
```

nia-e, 2023-05-16:
In this case you probably want
```rust
table.on_event(|mut table, event| match event {
    // Replace as necessary
    Event::DrawPartBegin => { /* ... */ }
    Event::DrawPartEnd => { /* ... */ }
})?;
```
These new variants on `Event` were implemented in the PR linked above and correspond to `LV_EVENT_DRAW_PART_BEGIN`, etc. See the PR for which exact variants were added

vortex314, 2023-05-16:
Looks I'm stuck to match C to Rust for the below  part. Do I need revert to lvgl_sys calls ? 
```

static void draw_part_event_cb(lv_event_t * e)
{
    lv_obj_t * obj = lv_event_get_target(e);
    lv_obj_draw_part_dsc_t * dsc = lv_event_get_draw_part_dsc(e);
    /*If the cells are drawn...*/
    if(dsc->part == LV_PART_ITEMS) {
        uint32_t row = dsc->id /  lv_table_get_col_cnt(obj);
        uint32_t col = dsc->id - row * lv_table_get_col_cnt(obj);

        /*Make the texts in the first cell center aligned*/
        if(row == 0) {
            dsc->label_dsc->align = LV_TEXT_ALIGN_CENTER;
            dsc->rect_dsc->bg_color = lv_color_mix(lv_palette_main(LV_PALETTE_BLUE), dsc->rect_dsc->bg_color, LV_OPA_20);
            dsc->rect_dsc->bg_opa = LV_OPA_COVER;
        }
        /*In the first column align the texts to the right*/
        else if(col == 0) {
            dsc->label_dsc->align = LV_TEXT_ALIGN_RIGHT;
        }

        /*MAke every 2nd row grayish*/
        if((row != 0 && row % 2) == 0) {
            dsc->rect_dsc->bg_color = lv_color_mix(lv_palette_main(LV_PALETTE_GREY), dsc->rect_dsc->bg_color, LV_OPA_10);
            dsc->rect_dsc->bg_opa = LV_OPA_COVER;
        }
    }
}
```

nia-e, 2023-05-16:
What specific functionality is missing? Can you file a separate issue so I can make a PR and link to it?

vortex314, 2023-05-16:
Ok, I will. Thanks.

PR #126 (2023-05-15): Handle more events
Closes #124