lvgl/lv_binding_rust #125: Cross compiling to ARM gives : unrecognized command-line option '-m64'

vortex314, 2023-05-10:
Trying to cross compile to ARMV6 for raspberry pi  fails at compiling lvgl_sys.
```
   Compiling anyhow v1.0.71
   Compiling time v0.1.45
   Compiling gtk v0.17.1
The following warnings were emitted during compilation:

warning: armv6-linux-musleabihf-gcc: error: unrecognized command-line option ‘-m64’

error: failed to run custom build command for `lvgl-sys v0.6.2 (https://github.com/lvgl/lv_binding_rust?branch=table-impl#1cc824c0)`

Caused by:
  process didn't exit successfully: `/home/lieven/workspace/lv_1/target/release/build/lvgl-sys-40c5db67c0326204/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-changed=/home/lieven/workspace/lv_1/./lv_conf.h
  cargo:rerun-if-changed=/home/lieven/workspace/lv_1/./lv_drv_conf.h
  cargo:rerun-if-env-changed=LVGL_INCLUDE
  cargo:rerun-if-env-changed=LVGL_LINK
  TARGET = Some("x86_64-unknown-linux-gnu")
  OPT_LEVEL = Some("0")
  HOST = Some("x86_64-unknown-linux-gnu")
  cargo:rerun-if-env-changed=CC_x86_64-unknown-linux-gnu
  CC_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CC_x86_64_unknown_linux_gnu
  CC_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CC
  HOST_CC = None
  cargo:rerun-if-env-changed=CC
  CC = Some("armv6-linux-musleabihf-gcc")
  cargo:rerun-if-env-changed=CFLAGS_x86_64-unknown-linux-gnu
  CFLAGS_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CFLAGS_x86_64_unknown_linux_gnu
  CFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CFLAGS
  HOST_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("false")
  CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
  running: "armv6-linux-musleabihf-gcc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-m64" "-I" "/home/lieven/.cargo/git/checkouts/lv_binding_rust-d86feb7597e107b7/1cc824c/lvgl-sys/vendor/lvgl/src" "-I" "/home/lieven/.cargo/git/checkouts/lv_binding_rust-d86feb7597e107b7/1cc824c/lvgl-sys/vendor" "-I" "/home/lieven/workspace/lv_1/." "-I" "/home/lieven/.cargo/git/checkouts/lv_binding_rust-d86feb7597e107b7/1cc824c/lvgl-sys/vendor/lv_drivers" "-I" "/usr/include/gtk-3.0" "-I" "/usr/include" "-I" "/usr/local/include" "-I" "/usr/include/glib-2.0" "-I" "/usr/lib/x86_64-linux-gnu/glib-2.0/include" "-I" "/usr/include/pango-1.0" "-I" "/usr/include/harfbuzz" "-I" "/usr/include/cairo" "-I" "/usr/include/gdk-pixbuf-2.0" "-I" "/usr/include/atk-1.0" "-I" "/usr/include/freetype2" "-DLV_CONF_INCLUDE_SIMPLE=1" "-o" "/home/lieven/workspace/lv_1/target/release/build/lvgl-sys-5243b0fad847d2b5/out/5f8af304d055c443-lv_draw_triangle.o" "-c" "/home/lieven/.cargo/git/checkouts/lv_binding_rust-d86feb7597e107b7/1cc824c/lvgl-sys/vendor/lvgl/src/draw/lv_draw_triangle.c"
  cargo:warning=armv6-linux-musleabihf-gcc: error: unrecognized command-line option ‘-m64’
  exit status: 1

  --- stderr
```

used this script ,  downloaded the cross-compiler to $HOME/toolchains
```
TARGET=arm-unknown-linux-gnueabi
TOOLCHAIN=armv6-linux-musleabihf
export PKG_CONFIG_PATH=$HOME/rpi-sysroot
export PKG_CONFIG_SYSROOT_DIR=$HOME/rpi-sysroot
export CC=$TOOLCHAIN-gcc
set -x
PATH="$HOME/toolchain/$TOOLCHAIN/bin:/usr/bin:/$HOME/.cargo/bin"
export CARGO_TARGET_ARM_UNKNOWN_LINUX_GNUEABI_LINKER=$TOOLCHAIN-gcc
cargo build --release --target $TARGET --verbose
```

nia-e, 2023-05-11:
Sorry for not getting to this; I'm a bit busy for now, I'll probably get around to this and your previous issue on Saturday. Sorry :c

vortex314, 2023-05-11:
No problemo. Ik saw other things i need you adapt, like sysroot, pkg-info  location.  Cross compilation is a very  complex story.

nia-e, 2023-05-11:
In the meantime, maybe worth checking with `cross`? #53 tackled that somewhat

vortex314, 2023-05-12:
Getting closer . Everything compiles, just the linking leads to a static vs dynamic conflict.
```
bject-2.0" "-lglib-2.0" "-lgio-2.0" "-lgobject-2.0" "-lglib-2.0" "-lcairo-gobject" "-lcairo" "-lgobject-2.0" "-lglib-2.0" "-lgobject-2.0" "-lglib-2.0" "-lgobject-2.0" "-lglib-2.0" "-Wl,--eh-frame-hdr" "-Wl,-z,noexecstack" "-L" "/home/lieven/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/arm-unknown-linux-gnueabihf/lib" "-o" "/home/lieven/workspace/lv_1/target/arm-unknown-linux-gnueabihf/release/deps/lv_1-f2f1b0e6c93976e8" "-Wl,--gc-sections" "-static" "-no-pie" "-Wl,-z,relro,-z,now" "-Wl,-O1" "-nodefaultlibs" "--sysroot=/home/lieven/rpi-sysroot" "-L/home/lieven/rpi-sysroot/usr/lib/arm-linux-gnueabihf"
  = note: /usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: /home/lieven/rpi-sysroot/usr/lib/arm-linux-gnueabihf/libc.a(strcpy.o): in function `strcpy':
          (.text+0x10): multiple definition of `strcpy'; /home/lieven/workspace/lv_1/target/arm-unknown-linux-gnueabihf/release/deps/liblvgl_sys-6ce0eba337f1ca7c.rlib(lvgl_sys-6ce0eba337f1ca7c.lvgl_sys.f53980ee-cgu.7.rcgu.o):lvgl_sys.f53980ee-cgu.7:(.text.strcpy+0x0): first defined here
          /usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: /home/lieven/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/arm-unknown-linux-gnueabihf/lib/libstd-1379e6211573016a.rlib(std-1379e6211573016a.std.18f4b1ee-cgu.0.rcgu.o): in function `std::sys::unix::os::home_dir::fallback':
          /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/sys/unix/os.rs:638: warning: Using 'getpwuid_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
          /usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: /home/lieven/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/arm-unknown-linux-gnueabihf/lib/libstd-1379e6211573016a.rlib(std-1379e6211573016a.std.18f4b1ee-cgu.0.rcgu.o): in function `<std::sys_common::net::LookupHost as core::convert::TryFrom<(&str,u16)>>::try_from::{{closure}}':
          /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/sys_common/net.rs:207: warning: Using 'getaddrinfo' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
          collect2: error: ld returned 1 exit status
          

error: could not compile `lv_1` due to previous error
```

vortex314, 2023-05-13:
What a difficult journey to get this working ! 
The program can compile now in 2 modes GTK on Linux AMD PC and cross compile to Raspberry Pi 1.
One of the struggles was the fact that some cross-compilation toolchains generate ARMV7 instructions while the Pi 1 only accepts ARMV6. 
There is one problem remaining the resolution looks very low and the colors are not correct. The hardware however is able to support the C version  https://github.com/lvgl/lv_port_linux_frame_buffer with correct colors.
On Raspberry Pi VGA it looks like this : 
[image]
The cross-compile script : https://github.com/vortex314/lv_1/blob/main/pi.sh

nia-e, 2023-05-13:
Bah, sorry I didn't come back to this - I've been swamped lately. I'm glad you got it working though; I assume at least some of the hacks in that script can be implemented in the `Cross.toml`. I'll do that as soon as I have a little bit of time

vortex314, 2023-05-13:
in the meantime, any idea why colors and resolution are off ?

nia-e, 2023-05-13:
The config file seems to be the same in both... maybe the resolution is for some reason set incorrectly? Given that it's stretched, my guess is that for some reason the framebuffer device is drawing off-screen (given that it's CRT, maybe it's drawing in the HBLANK/VBLANK?) but it's odd that the C version works fine. I'll look into this more

vortex314, 2023-05-13:
Ok, has been solved. The color depth needs to be 16 , not 32 on this raspberry pi. Changed the lv_conf.h accordingly. 
The fbdev_init()  example extracted this from the /dev/fb0 itself .  But was incorrect at an higher layer by lv_conf.h 
```
[Info]  (0.000, +0)      fbdev_init: The framebuffer device was opened successfully     (in fbdev.c line #99)
[Info]  (0.000, +0)      fbdev_init: 1024x768, 16bpp    (in fbdev.c line #147)
[Info]  (0.000, +0)      fbdev_init: The framebuffer device was mapped to memory successfully   (in fbdev.c line #162)
```