lvgl/lv_binding_rust #127: Table first row color different than other rows.

vortex314, 2023-05-16:
Just typed a full description and then Github gave an error 500. Ok. Here we go again.
Basically target is the same as this : https://docs.lvgl.io/master/widgets/table.html#simple-table
Event class is missing some functions : get_target(), get_draw_part().
Underlying the drawpart there are structures that enable to change cell rectangles and labels. For which no Rust equivalent has been defined so far. 
Can this be exposed ?

nia-e, 2023-05-18:
I've been trying to figure out a way to do this safely for a while but I honestly have no idea. I think for now using the `lvgl_sys` unsafe bindings is needed; I just don't see a way to be able to do this safely easily. There probably is a way but I worry about `get_target()` specifically since it can give multiple `Obj`ects which all point to the same `lv_obj_t` and thus can cause race conditions.

vortex314, 2023-05-21:
Ok, I see , thanks for the feedback. I'll see if using lvgl_sys is workable.

closed 2023-05-21 by vortex314