lvgl/lv_binding_rust #147: pub struct DrawBuffer<const N: usize>

fulup-bzh, 2023-08-04:
Forcing this to a constant prevent from having simulator screen size store in a config object.
Is there any special reason to enforce const here ?

fulup-bzh, 2023-08-04:
I'm not sure to understand, this is an integer and not a pointer it is copies once also the CONTS does not provide any functionnality

nia-e, 2023-08-04:
The `const N` is used to determine the size of the allocation for the buffer; we store a `Pin<Box<[MaybeUninit<lvgl_sys::lv_color_t>; N]>>` inside of the actual `DrawBuffer` struct (note that this is not the same `Box` as in the rust stdlib; we use our own implementation, see [mem.rs](https://github.com/lvgl/lv_binding_rust/blob/master/lvgl/src/mem.rs)). Because the innermost value is a sized slice `[T; N]` we can't actually allocate it at runtime; we'd need a `Vec<T>` impl or similar, which is considerably more complex than just our boutique `Box<T>`. Arguably this is a Rust limitation, but short of doing some *extremely* unsafe things I don't see how to get around it short of said custom `Vec` implementation

fulup-bzh, 2023-08-04:
I understand your point, it is a before everything a Rust limitation. Where you cannot force a specific element of a structure to be constance. If is was a string, we could use ```&'static str``` for integer it does not exist. 

The point I do not get is that for me they is not difference in between the two following options
* DrawBuffer (255*128)
* sz= config.h*config.v; DrawBuffer(sz)

nia-e, 2023-08-04:
In the latter case, you're getting the values at runtime (since the compiler doesn't know the value of `config.h` or of `config.v`) so the actual size of the final `DrawBuffer` isn't known by the compiler. You could work around this with a build script that sets some constants, or by using `const fn`s to calculate the size (if the config-reading crate you use supports it).

fulup-bzh, 2023-08-04:
I understand your point. I will try to move around by checking if a ```const fn```

closed 2023-08-04 by fulup-bzh