lvgl/lv_binding_rust #148: Isolate build display from update (DrawBuffer ???)

fulup-bzh, 2023-08-07:
I run an asynchronous application, where display construction if done at init time and update from an independant asynchronous timer. I started from ```arc``` sample implementing the construction in one routine and display update in an other. When trying to update the window I get a core dump, I suspect ```DrawBuffer``` or an other holding handle to be wipe out when returning from init routine. 

Question: which are the handle I should keep track of to make sure that my display does not get wipe out from stack when returning from my init routine ? (using arc sample as base)

Note:  In order to keep different display handles from being wipe out at return my routine returns. 
```
       return DisplayHandle {
           sim_display,
            window,
            arc,
            display,
        }
```
I cannot return buffer coming from DrawBuffer because it of its type that depends in the definition.

My timer async routine fail in lvgl::task_handler() 
```
  pub fn tick_update(&mut self, tick: Duration) {

        // update internal lvgl tick timer
        lvgl::tick_inc(tick);

        // exec any lvgl pending task/action
        lvgl::task_handler();
}
```

fulup-bzh, 2023-08-07:
For information debugger trace.
[image]

nia-e, 2023-08-08:
Are you moving the actual display between threads? If so, handling this case is still an open question; see https://github.com/lvgl/lv_binding_rust/issues/140. I was trying to partially fix this in https://github.com/lvgl/lv_binding_rust/pull/144/ but I haven't had time to work on this project recently

fulup-bzh, 2023-08-08:
Potential display widgets may may change value/status from a different thread. Nevertheless the update action handle from a timer always come from the same thread.
The issue seems more to be that when returning from my create function, some LVGL handle are free. I'm expecially afrby Drawbuffer could reserve memory in stack and not in a Box::leak

fulup-bzh, 2023-08-08:
Digging further, I found a warning that alarm me.
```
```

nia-e, 2023-08-08:
I believe that has since been patched, it should be fine on latest git master?

fulup-bzh, 2023-08-09:
I updated to master without more success. To make this clearer to follow I propose to close this thread and reopen a new cleaner one on "updating display from an independent context"

closed 2023-08-09 by fulup-bzh