lvgl/lv_binding_rust #152: compile error - error: failed to run custom build command for `lvgl-sys v0.6.2

ds2k5, 2023-11-02:
Hi,
here is what I did

```
$ git clone https://github.com/lvgl/lv_binding_rust
$ cd lv_binding_rust
$ . ~/export-esp.sh
$ export DEP_LV_CONFIG_PATH=~/lv_binding_rust/examples/include
$ cargo check

```
```
error: failed to run custom build command for `lvgl-sys v0.6.2 (/home/developer/rust/lv_binding_rust/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/home/developer/lv_binding_rust/target/debug/build/lvgl-sys-d765cd76abd41e43/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at lvgl-sys/build.rs:95:13:
  Directory /home/developer/lv_binding_rust/examples referenced by DEP_LV_CONFIG_PATH needs to contain a file called lv_conf.h
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```

Ddystopia, 2024-04-19:
I am having the same issue. Did you resolve it?

ds2k5, 2024-04-19:
Sorry no did no solved it...

```
   Compiling lvgl-sys v0.6.2 (/home/developer/lv_binding_rust/lvgl-sys)
error: failed to run custom build command for `lvgl-sys v0.6.2 (/home/developer/lv_binding_rust/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/home/developer/lv_binding_rust/target/debug/build/lvgl-sys-7bc7e4fb1b99cbd3/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=/home/developer/lv_binding_rust/fonts
  cargo:rerun-if-changed=/home/developer/lv_binding_rust/examples/include/lv_conf.h
  cargo:rerun-if-changed=/home/developer/lv_binding_rust/examples/include/lv_drv_conf.h
  cargo:rerun-if-env-changed=LVGL_INCLUDE
  cargo:rerun-if-env-changed=LVGL_LINK

  --- stderr
  thread 'main' panicked at lvgl-sys/build.rs:269:39:
  called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```

ds2k5, 2024-10-12:
@Ddystopia 

did you found a solution ?

Ddystopia, 2024-10-12:
Yeah, I followed steps in readme. 

```

$ git submodule init
$ git submodule update 

```

After using that library, we actually kind of regret it. It is full of UB, terrible, lifetimes are wrong, abstractions are strange, functionality is missing etc.

It would've been better to just write our own wrappers tbh.

ds2k5, 2024-10-12:
@Ddystopia 

did not work for me

git submodule init
git submodule update 
export DEP_LV_CONFIG_PATH=~/rust/lv_binding_rust/examples/include
cargo check

```
   Compiling ctor v0.2.8
   Compiling lvgl-sys v0.6.2 (/home/developer/rust/lv_binding_rust/lvgl-sys)
The following warnings were emitted during compilation:

warning: lvgl-sys@0.6.2: <command-line>: error: no macro name given in #define directive

error: failed to run custom build command for `lvgl-sys v0.6.2 (/home/developer/rust/lv_binding_rust/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/home/developer/rust/lv_binding_rust/target/debug/build/lvgl-sys-94ff0e8720750954/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-changed=/home/developer/rust/lv_binding_rust/fonts
  cargo:rerun-if-changed=/home/developer/rust/lv_binding_rust/examples/include/lv_conf.h
  cargo:rerun-if-changed=/home/developer/rust/lv_binding_rust/examples/include/lv_drv_conf.h
  cargo:rerun-if-env-changed=LVGL_INCLUDE
  cargo:rerun-if-env-changed=LVGL_LINK
  OUT_DIR = Some(/home/developer/rust/lv_binding_rust/target/debug/build/lvgl-sys-15d09bab13537cfa/out)
  TARGET = Some(x86_64-unknown-linux-gnu)
  OPT_LEVEL = Some(0)
  HOST = Some(x86_64-unknown-linux-gnu)
  cargo:rerun-if-env-changed=CC_x86_64-unknown-linux-gnu
  CC_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CC_x86_64_unknown_linux_gnu
  CC_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CC
  HOST_CC = None
  cargo:rerun-if-env-changed=CC
  CC = None
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  RUSTC_WRAPPER = None
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some(true)
  CARGO_CFG_TARGET_FEATURE = Some(fxsr,sse,sse2)
  cargo:rerun-if-env-changed=CFLAGS_x86_64-unknown-linux-gnu
  CFLAGS_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CFLAGS_x86_64_unknown_linux_gnu
  CFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CFLAGS
  HOST_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None
  cargo:warning=<command-line>: error: no macro name given in #define directive

  --- stderr

  error occurred: Command "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-gdwarf-4" "-fno-omit-frame-pointer" "-m64" "-I" "/home/developer/rust/lv_binding_rust/lvgl-sys/vendor/lvgl/src" "-I" "/home/developer/rust/lv_binding_rust/lvgl-sys/vendor" "-I" "/home/developer/rust/lv_binding_rust/examples/include" "-I" "/home/developer/rust/lv_binding_rust/fonts" "-I" "/home/developer/rust/lv_binding_rust/lvgl-sys/vendor/lv_drivers" "-I" "/usr/include" "-I" "/usr/local/include" "-DLV_CONF_INCLUDE_SIMPLE=1" "-D=" "-o" "/home/developer/rust/lv_binding_rust/target/debug/build/lvgl-sys-15d09bab13537cfa/out/cae12387a3ab71a0-fonts_noto_sans_numeric_80.o" "-c" "/home/developer/rust/lv_binding_rust/fonts/fonts_noto_sans_numeric_80.c" with args cc did not execute successfully (status code exit status: 1).

```

Tathar, 2024-10-18:
you can use 

[patch.crates-io]
lvgl = { git = "https://github.com/enelson1001/lv_binding_rust"}
lvgl-sys = { git = "https://github.com/enelson1001/lv_binding_rust"}

in Cargo.toml

Find here "https://github.com/enelson1001/rust-esp32s3-lvgl-clickme"

ds2k5, 2024-10-19:
$ cd lv_binding_rust

cat Cargo.toml

```

[workspace]
members = [
    "lvgl",
    "lvgl-codegen",
    "lvgl-sys",
]

resolver = "2"

[patch.crates-io]
lvgl = { git = "https://github.com/enelson1001/lv_binding_rust"}
lvgl-sys = { git = "https://github.com/enelson1001/lv_binding_rust"}

```

$ cargo clean
$ . ~/export-esp.sh
$ export DEP_LV_CONFIG_PATH=~/lv_binding_rust/examples/include
$ cargo check

```
 Compiling lvgl-codegen v0.6.2 (/home/developer/rust/lv_binding_rust/lvgl-codegen)
   Compiling ctor v0.2.8
   Compiling lvgl-sys v0.6.2 (/home/developer/rust/lv_binding_rust/lvgl-sys)
error: failed to run custom build command for `lvgl-sys v0.6.2 (/home/developer/rust/lv_binding_rust/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/home/developer/rust/lv_binding_rust/target/debug/build/lvgl-sys-94ff0e8720750954/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at lvgl-sys/build.rs:89:13:
  Directory /home/developer/lv_binding_rust/examples/include referenced by DEP_LV_CONFIG_PATH needs to exist
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```

Tathar, 2024-10-19:
@ds2k5 

After checking it seems that the "~/lv_binding_rust/examples/include" folder does not exist 

>>Directory /home/developer/lv_binding_rust/examples/include referenced by DEP_LV_CONFIG_PATH needs to exist

ds2k5, 2024-10-19:
$ ls ~/rust/lv_binding_rust/examples/include
lv_conf.h  lv_drv_conf.h

Tathar, 2024-10-21:
@ds2k5

~/rust/lv_binding_rust/examples/include
is not
~/lv_binding_rust/examples/include

ds2k5, 2024-10-22:
@Tathar 

you are right - it was a typo from my side

Tathar, 2024-10-22:
@ds2k5

can you post your your "cargo.toml" and ".cargo/config.toml" ?

ds2k5, 2024-10-23:
@Tathar 

Cargo.toml

```
[workspace]
members = [
    "lvgl",
    "lvgl-codegen",
    "lvgl-sys",
]

resolver = "2"

[patch.crates-io]
lvgl = { git = "https://github.com/enelson1001/lv_binding_rust"}
lvgl-sys = { git = "https://github.com/enelson1001/lv_binding_rust"}

```

.cargo/config.toml
```

[target.aarch64-apple-darwin]
linker = "/home/developer/osxcross/target/bin/aarch64-apple-darwin20.4-clang"
ar = "/home/developer/osxcross/target/bin/aarch64-apple-darwin20.4-ar"
[target.x86_64-apple-darwin]
linker = "/home/developer/osxcross/target/bin/o64-gcc"
[target.armv7-unknown-linux-gnueabihf]
linker = "/usr/bin/arm-linux-gnueabihf-gcc"
runner = "/usr/bin/qemu-arm"
[target.i686-unknown-linux-gnu]
linker = "/usr/bin/i686-linux-gnu-gcc"
[target.xtensa-esp32-none-elf]
linker = "/home/developer/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"
[target.xtensa-esp32-espidf]
linker = "/home/developer/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"
[target.thumbv6m-none-eabi]
runner = "elf2uf2-rs -d"

```

Tathar, 2024-10-23:
need add target into .cargo/config.toml (in first line)

[build]
target = "xtensa-esp32-espidf"

for exemple

ds2k5, 2024-10-23:
>need add target into .cargo/config.toml (in first line)

I did but ...

```
$ cargo check
warning: Patch `lvgl v0.6.2 (https://github.com/enelson1001/lv_binding_rust#723ad383)` was not used in the crate graph.
Perhaps you misspelled the source URL being patched.
Possible URLs for `[patch.<URL>]`:
    /home/developer/rust/lv_binding_rust/lvgl
warning: Patch `lvgl-sys v0.6.2 (https://github.com/enelson1001/lv_binding_rust#723ad383)` was not used in the crate graph.
Perhaps you misspelled the source URL being patched.
Possible URLs for `[patch.<URL>]`:
    /home/developer/rust/lv_binding_rust/lvgl-sys
   Compiling regex-automata v0.4.8
   Compiling prettyplease v0.2.22
   Compiling which v4.4.2
   Compiling cexpr v0.6.0
    Checking memchr v2.7.4
   Compiling autocfg v1.4.0
   Compiling az v1.2.1
   Compiling proc-macro2 v1.0.87
'esp32' is not a recognized processor for this target (ignoring processor)
'esp32' is not a recognized processor for this target (ignoring processor)
'esp32' is not a recognized processor for this target (ignoring processor)
'esp32' is not a recognized processor for this target (ignoring processor)
error[E0463]: can't find crate for `core`
  |
  = note: the `xtensa-esp32-espidf` target may not be installed
  = help: consider downloading the target with `rustup target add xtensa-esp32-espidf`

   Compiling cty v0.2.2
For more information about this error, try `rustc --explain E0463`.
error: could not compile `memchr` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
developer@w541:~/rust/lv_binding_rust$ 
developer@w541:~/rust/lv_binding_rust$ 
developer@w541:~/rust/lv_binding_rust$ cargo clean
     Removed 335 files, 223.2MiB total
developer@w541:~/rust/lv_binding_rust$ 
developer@w541:~/rust/lv_binding_rust$ 
developer@w541:~/rust/lv_binding_rust$ 
developer@w541:~/rust/lv_binding_rust$ cargo check
warning: Patch `lvgl v0.6.2 (https://github.com/enelson1001/lv_binding_rust#723ad383)` was not used in the crate graph.
Perhaps you misspelled the source URL being patched.
Possible URLs for `[patch.<URL>]`:
    /home/developer/rust/lv_binding_rust/lvgl
warning: Patch `lvgl-sys v0.6.2 (https://github.com/enelson1001/lv_binding_rust#723ad383)` was not used in the crate graph.
Perhaps you misspelled the source URL being patched.
Possible URLs for `[patch.<URL>]`:
    /home/developer/rust/lv_binding_rust/lvgl-sys
   Compiling proc-macro2 v1.0.87
   Compiling memchr v2.7.4
   Compiling unicode-ident v1.0.13
   Compiling glob v0.3.1
   Compiling regex-syntax v0.8.5
   Compiling prettyplease v0.2.22
   Compiling libc v0.2.159
   Compiling rustix v0.38.37
   Compiling linux-raw-sys v0.4.14
   Compiling bitflags v2.6.0
   Compiling minimal-lexical v0.2.1
   Compiling lazy_static v1.5.0
   Compiling aho-corasick v1.1.3
   Compiling cfg-if v1.0.0
   Compiling libloading v0.8.5
   Compiling nom v7.1.3
   Compiling clang-sys v1.8.1
   Compiling quote v1.0.37
   Compiling bindgen v0.65.1
   Compiling syn v2.0.79
   Compiling shlex v1.3.0
   Compiling either v1.13.0
   Compiling home v0.5.9
   Compiling bitflags v1.3.2
   Compiling lazycell v1.3.0
   Compiling peeking_take_while v0.1.2
   Compiling log v0.4.22
   Compiling rustc-hash v1.1.0
   Compiling cc v1.1.30
   Compiling regex-automata v0.4.8
   Compiling cexpr v0.6.0
   Compiling which v4.4.2
'esp32' is not a recognized processor for this target (ignoring processor)
'esp32' is not a recognized processor for this target (ignoring processor)
'esp32' is not a recognized processor for this target (ignoring processor)
'esp32' is not a recognized processor for this target (ignoring processor)
   Compiling autocfg v1.4.0
error[E0463]: can't find crate for `core`
  |
  = note: the `xtensa-esp32-espidf` target may not be installed
  = help: consider downloading the target with `rustup target add xtensa-esp32-espidf`

For more information about this error, try `rustc --explain E0463`.
error: could not compile `memchr` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...

```