lvgl/lv_binding_rust #158: Alternative option for Rust/Lvgl binding

fulup-bzh, 2023-12-22:
After trying to use lv_binding_rust for almost two weeks, we finally gave up. We add two many memory issue and we came to the conclusion that fixing memory model would require more work than rewriting a new lvgl/rust binding based on simpler concepts. *Note we are working in Cross environment, targeting a Arm platform in frame-buffer and or goal was to publish a simple UI for an EV-Charger, also we need only a subset of LVGL features/options.*

* Our work is available at https://github.com/tux-evse/lvgl-rclib-rs * : feel free to take any idea you may find useful. 

**Few lesson learn from of lvgl/rust work:**
* leaking widget memory structure is far simpler and match pretty well embedded constrains, where widget are usually static and not create on the fly.
* proposing a hight level API, somehow like Phyton does, make developer far simpler. It also simplyfy the binding as many method are common to all widget (ex: lock, border, size, ...)
* implementing a "super class" model for widget allow to manipulate any widget in a generic way without dealing with each specific flavour (Label, Button, ...). This is especially important when retrieving/sending event from en external mainloop.
* Reducing dependencies: we have to maintain our code for 10-15 years and limiting external dependencies to the strict minimum is a MUST have feature.

LVGL is really a very cool library, and having a higher API model make easier for our graphic developer that are not C/C++ expert to development nice UI on embedded device.

C47D, 2025-01-09:
> Personally I've found it easier to just interface directly with the raw LVGL functions than to try to use this Rust wrapper

Do you have any advice on how to improve this wrapper? We're starting to work on it again and me personally I'm working on this because I have interest in rust lang, so any advice is welcome

C47D, 2025-01-09:
Hi, thanks again for taking some time to give us feedback, this repo went silent for a while and we're catching up.

---
We're currently discussing separating `lvgl-sys` into another repo so it can be updated when LVGL releases a new version (major, minor or patches), see the discussion here #176 

---
I've added the issues you listed in this other issue #184 were we have some action points for this repo. And about the wrapper I've just opened a PR adding some methods/traits to Label and Textarea, do you mean that kind of wrappers? See #188 , I'm just getting familiar with Rust and this repo

fragsalat, 2025-09-24:
> Personally I've found it easier to just interface directly with the raw LVGL functions than to try to use this Rust wrapper

Holy moly I tried days and hours to get this library work and 1 minute with the ask to copilot to get rid of this lvgl-rs wrapper and interface the clibrary directly and it works like a charm. There is no way to get this library working on windows or linux imo. Even when it compiles I get pointer exceptions even with the basic examples. 

Asking copilot just created this

<details>
<summary>lvgl_direct.rs</summary>

```
    use std::ffi::{CStr, CString, c_void};
    use std::os::raw::{c_char, c_int, c_uint};
    use std::ptr;
    use std::thread::sleep;
    use std::time::Duration;
    
    use embedded_graphics::pixelcolor::Rgb565;
    use embedded_graphics::prelude::*;
    use embedded_graphics_simulator::{
        OutputSettingsBuilder, SimulatorDisplay, SimulatorEvent, Window,
    };
    
    // Direct LVGL C bindings
    #[link(name = "lvgl")]
    extern "C" {
        fn lv_init();
        fn lv_task_handler();
        fn lv_tick_inc(tick_period: c_uint);
    
        // Display functions
        fn lv_disp_create(hor_res: c_int, ver_res: c_int) -> *mut c_void;
        fn lv_disp_set_draw_buf(disp: *mut c_void, buf: *mut c_void, size: c_uint);
        fn lv_disp_set_flush_cb(
            disp: *mut c_void,
            flush_cb: extern "C" fn(*mut c_void, *const c_void, *const c_void),
        );
    
        // Screen/object functions
        fn lv_scr_act() -> *mut c_void;
        fn lv_obj_create(parent: *mut c_void) -> *mut c_void;
        fn lv_obj_set_size(obj: *mut c_void, width: c_int, height: c_int);
        fn lv_obj_set_pos(obj: *mut c_void, x: c_int, y: c_int);
        fn lv_obj_set_style_bg_color(obj: *mut c_void, color: c_uint, selector: c_uint);
    
        // Button functions
        fn lv_btn_create(parent: *mut c_void) -> *mut c_void;
    
        // Input device functions
        fn lv_indev_create() -> *mut c_void;
        fn lv_indev_set_type(indev: *mut c_void, indev_type: c_int);
        fn lv_indev_set_read_cb(indev: *mut c_void, read_cb: extern "C" fn(*mut c_void, *mut c_void));
    }
    
    // LVGL constants
    const LV_INDEV_TYPE_POINTER: c_int = 0;
    const LV_PART_MAIN: c_uint = 0;
    
    // Color helper
    fn lv_color_make(r: u8, g: u8, b: u8) -> c_uint {
        ((r as c_uint) << 16) | ((g as c_uint) << 8) | (b as c_uint)
    }
    
    // Input data structure (simplified)
    #[repr(C)]
    struct LvIndevData {
        point_x: c_int,
        point_y: c_int,
        state: c_int, // 0 = released, 1 = pressed
    }
    
[... 351 lines cut ...]
                                    break 'running;
                                }
                                _ => {
                                    println!("Button {} action triggered", button_id);
                                }
                            }
                        }
                    }
                    SimulatorEvent::Quit => break 'running,
                    _ => {}
                }
            }
    
            sleep(Duration::from_millis(16)); // ~60 FPS
        }
    
        println!("Application terminated successfully.");
        Ok(())
    }

```
</details>

While this is just hacked together with copilot as quick and dirty solution it took 1 minute and works!
So to the maintainer, truly thanks for your effort and the nice foundational work, I would have wished it worked. But if you have no time to maintain this library, better mark it as out of date and don't recommend it to be used :)