lvgl/lv_binding_rust #168: Build error when using the git repo in Cargo.toml

amcelroy, 2024-05-16:
Hi,

I'm trying to use the latest version of the repo that uses `embedded-graphics` version 0.8.0. The crates.io version is using 0.7.1 and is causing dependency collisions with a display driver. The `Cargo.toml` for the project looks like:

```
[dependencies.lvgl]
# path = "./lv_binding_rust/lvgl"
# version = "0.6.2"
git = "https://github.com/lvgl/lv_binding_rust"
default-features = false
features = ["embedded_graphics", "unsafe_no_autoinit", "lvgl_alloc"]
```

However, the following error is generated:
```
error[E0599]: no method named `into_raw` found for struct `Box<lv_style_t>` in the current scope
   --> /Users/myname/.cargo/git/checkouts/lv_binding_rust-d86feb7597e107b7/45cafea/lvgl/src/lv_core/style.rs:459:58
    |
459 |             lvgl_sys::lv_style_get_prop(self.raw.clone().into_raw() as *const _, prop.bits(), ptr)
    |                                         -----------------^^^^^^^^--
    |                                         |                |
    |                                         |                this is an associated function, not a method
    |                                         help: use associated function syntax instead: `Box::<lv_style_t>::into_raw(self.raw.clone())`
    |
    = note: found the following associated functions; to be used as methods, functions must have a `self` parameter
    = note: the candidate is defined in an impl for the type `Box<T, A>`

For more information about this error, try `rustc --explain E0599`.
error: could not compile `lvgl` (lib) due to 1 previous error
make: *** [build] Error 101
```

CFSworks, 2024-10-23:
I have just encountered this issue as well.

The problem is that the `lvgl` crate includes its own implementation of `Box` (which represents memory allocations managed by LVGL). This implementation includes into_raw() [as an instance method](https://github.com/lvgl/lv_binding_rust/blob/3b3abb3019eef637b932fae967a8d246bf5b6f14/lvgl/src/mem.rs#L39) (with `self`), contrary to Rust's implementation [as an associated function](https://doc.rust-lang.org/std/boxed/struct.Box.html#method.into_raw) (without `self`) and thus must be used as `Box::into_raw(...)`. This is intentionally done by the language designers to avoid confusion in cases where the inner type might define "into_raw" as an instance method of its own.

When the `lvgl_alloc` feature is activated, the LVGL memory management code takes over as *the* global allocator. This makes the LVGL custom `Box` type redundant since *all* boxes are in LVGL's heap already. So [LVGL just switches over to the standard `alloc` implementation instead](https://github.com/lvgl/lv_binding_rust/blob/3b3abb3019eef637b932fae967a8d246bf5b6f14/lvgl/src/lib.rs#L34). But .into_raw() is not the correct way to access *real* boxes, leading to this error.

To resolve this issue, the LVGL box should be updated to match the Rust convention of making `into_raw` an associated function, not an instance method. Eventually, when the `allocator_api` language feature is stabilized, the core `Box` type will allow boxes to live in allocators besides the global one, and LVGL should remove its bespoke `Box` type in favor of this new mechanism.

C47D, 2025-01-09:
@CFSworks Thanks for pointing out the solution! We'll work on it.