lvgl/lv_binding_rust #176: Put lvgl-sys into another repository

AlixANNERAUD, 2024-07-01:
Since `lv_binding_rust` appears unlikely to be updated for LVGL v9.X anytime soon, could `lvgl-sys` be moved to a separate repository to allow for regular updates independently of `lv_binding_rust`/`lvgl`?

AlixANNERAUD, 2024-12-19:
I already have a [repository ready](https://github.com/Xila-Project/lvgl_rust_sys), we just need to change ownership and add automatic release for previous versions (down to v8.0.0 ? maybe).

C47D, 2024-12-27:
Hi,

If a user wants to use the lvgl rust bindings what should they include, this repo or the `lvgl-sys` repo?

I have the impression that this repo is more an examples repo and moving `lvgl-sys` to a separated repo sound like a good thing to do. I guess we should also need update it when main lvgl make a new release. Am I missing something else?

Any thoughts @kisvegabor?

AlixANNERAUD, 2024-12-27:
Regular end users should rely on the safe Rust bindings. However, advanced users might choose to use the unsafe raw bindings (via the lvgl-sys C library generated by bindgen), either as a complement or directly, depending on their needs.

The release schedule for the repositories should be as follows:
	
- Raw (lvgl-sys): Automatically updated whenever a new version of LVGL is released (major, minor, or patch).
- Safe bindings: Released when ready, typically aligned with major LVGL versions. The lvgl-sys library should maintain compatibility with any minor or patch updates, as these do not introduce breaking API changes.

C47D, 2024-12-28:
> Regular end users should rely on the safe Rust bindings. However, advanced users might choose to use the unsafe raw bindings (via the `lvgl-sys` C library generated by `bindgen`), either as a complement (to use API unsupported by the safe one) or directly, depending on their needs.
> 
> The release schedule for the repositories should be as follows:
> 	
> - Raw (`lvgl-sys`): Automatically (CI) released whenever a new version of LVGL is released (major, minor, or patch).
> - Safe bindings: Released when ready, typically aligned with major LVGL versions. The `lvgl-sys` dependency version should be set to any minor or patch updates, as these do not introduce breaking API changes, thus no change in the safe part code generator.
> 
> So you are mostly right, I added that for precision purpose.

And the safe bindings are this repo? They have to be done manually? Sorry for the dumb questions tho

AlixANNERAUD, 2024-12-28:
Don't worry. This repository currently includes both the safe and unsafe bindings. If we separate out the unsafe bindings, what remains will be the safe bindings (and their generator library).

kisvegabor, 2025-01-03:
It might be a little off topic, but also related to the schedule of the releases and stability. In the the lvgl repo we have a test which should be added for all bindings. [This UI](https://github.com/lvgl/lvgl/blob/master/tests/ref_imgs/binding.png) needs to be created. [This is the C](https://github.com/lvgl/lvgl/blob/master/tests/src/test_cases/test_bindings.c) code. For MicroPython we have [the equivalent test](https://github.com/lvgl/lvgl/blob/master/tests/micropy_test/micropy.py) which uses the very same reference image. Both test are part of LVGL's CI so we test MicroPython on each commit. 

In order to support Rust in a serious level we need to add a similar test to make sure that the binding is not broken. 

What do you think about it?

C47D, 2025-01-03:
> It might be a little off topic, but also related to the schedule of the releases and stability. In the the lvgl repo we have a test which should be added for all bindings. [This [...]

Sounds like a good first step to keep this binding working. I will travel tonight back to home, so I can work on this this weekend

AlixANNERAUD, 2025-01-04:
> It might be a little off topic, but also related to the schedule of the releases and stability. In the the lvgl repo we have a test which should be added for all bindings. [This [...]

Yeah that makes sense and it is easy to implement.

C47D, 2025-01-08:
@kisvegabor With regards of separating the `lvgl-sys` repo from this, do you agree, see any issue?

kisvegabor, 2025-01-09:
> However there is some stuff like img in this test that wasn't supported before so I don't expect it to fully pass anytime soon.

That binding test is quite complete. If he binding is capable of doing all these, most of the people should be happy with it. So I think it's also good measure for readiness. 

> With regards of separating the lvgl-sys repo from this, do you agree, see any issue?

Basically there would be 2 bindings: a safe and an unsafe? If so, I suggest focusing on the simpler first, and if it ready, working, and we have positive feedback, we can create a repo for the other too. What do you think?

C47D, 2025-01-09:
> Basically there would be 2 bindings: a safe and an unsafe? If so, I suggest focusing on the simpler first, and if it ready, working, and we have positive feedback, we can create a repo for the other too. What do you think?

Correct, the unsafe bindings are created automatically, but having `lvgl-sys` in a separated repo would allow us to generate unsafe bindings for multiple LVGL versions. Currently the unsafe bindings available in this repo are 'stuck' in LVGLv8.

See this another comment about it https://github.com/lvgl/lv_binding_rust/issues/158#issuecomment-2579169093

kisvegabor, 2025-01-13:
What about using branches like `release/v8.3`, `release/v8.4`, `release/v9.0`, etc?
Same as we do in the `lvgl` repo?

AlixANNERAUD, 2025-01-14:
> What about using branches like `release/v8.3`, `release/v8.4`, `release/v9.0`, etc? Same as we do in the `lvgl` repo?

Exactly

C47D, 2025-01-14:
Ok then, I'll create a new branch with the current version and then move the lvgl submodule to v9, then we can start patching it as @AlixANNERAUD did in their repo.
Does it sounds ok?

C47D, 2025-01-15:
I've just created the `release/v8.3 branch, does it sounds ok?
[image]

AlixANNERAUD, 2025-01-15:
> I've just created the `release/v8.3` branch, does it sounds ok? If so I can create new branches with all the other releases taking the main LVGL repo as reference.
> 
> 
> 
> [image]
> 
> 
> 

In the current repository ?

C47D, 2025-01-16:
I think @kisvegabor is more in favor of having it in the same repo. Having `lvgl-sys` (a.k.a. FFI bindings) will allow us to include them without all the other bindings, isn't?

AlixANNERAUD, 2025-01-16:
The main point of my original issue is that moving `lvgl-sys` to a separate repository would give users the flexibility to choose between the unsafe bindings or the full-fledged safe bindings in Rust.

The current configuration is impractical and confusing, as it requires maintaining a compatible version of the safe `lvgl` bindings for each version of the `lvgl-sys` bindings.

Decoupling the two would simplify this process, allowing users to choose more easily and enabling `lvgl-sys` bindings to be released promptly whenever a new LVGL version is available since it is automatically generated and would not require adaptation for any future release (since raw C to rust bindings generation are stable), which would not be the case for the safe bindings due to its complexity.

This approach mirrors the structure of the LittleFS Rust bindings, where the [[safe bindings](https://github.com/trussed-dev/littlefs2)](https://github.com/trussed-dev/littlefs2) are hosted in one repository and the [[unsafe bindings](https://github.com/trussed-dev/littlefs2-sys)](https://github.com/trussed-dev/littlefs2-sys) in another, providing a clear and modular separation.

C47D, 2025-01-22:
Any updates @kisvegabor ? I see the LittleFS repo as a good example

kisvegabor, 2025-02-06:
Sorry for the delay. 🙁 

I'm ok with having 2 repos for the safe and unsafe binding as it seems they are really separate things (different code base, different issues, different versioning). However for versions I think we should use the branches as it was proposed by @C47D I can't comment on the Rust related details , but having a repo for each version would make the lvgl GitHub organization nosiy.

Jiiua, 2025-10-04:
I can't understand why you are saying "noisy" @kisvegabor, i have looked into the organization and its already pretty much exactly noisy, i have no clue what doing this is changing, seems just like moving goal posts.

It has been several months, maintainers are waiting to actually do work, don't block the flow.

> Sorry for the delay. 🙁
>
> I'm ok with having 2 repos for the safe and unsafe binding as it seems they are really separate things (different code base, different issues, different versioning). However [...]