lvgl/lv_binding_rust #178: Fail

AlixANNERAUD, 2024-07-18:
Hello, apart from the latest commits that have completely broken the compilation, I am unable to compile lvgl : 
```
error: failed to run custom build command for `lvgl-sys v0.6.2 (/home/alix_anneraud/Git/Personnel/Xila/lv_binding_rust/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/home/alix_anneraud/Git/Personnel/Xila/Core/target/debug/build/lvgl-sys-d94f608824b656ec/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=/home/alix_anneraud/Git/Personnel/Xila/Core/Modules/Graphics/include/lv_conf.h
  cargo:rerun-if-changed=/home/alix_anneraud/Git/Personnel/Xila/Core/Modules/Graphics/include/lv_drv_conf.h
  cargo:rerun-if-env-changed=LVGL_INCLUDE
  cargo:rerun-if-env-changed=LVGL_LINK
  OUT_DIR = Some(/home/alix_anneraud/Git/Personnel/Xila/Core/target/debug/build/lvgl-sys-2ac71f54f14e97e9/out)
  TARGET = Some(x86_64-unknown-linux-gnu)
  OPT_LEVEL = Some(0)
  HOST = Some(x86_64-unknown-linux-gnu)
  cargo:rerun-if-env-changed=CC_x86_64-unknown-linux-gnu
  CC_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CC_x86_64_unknown_linux_gnu
  CC_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CC
  HOST_CC = None
  cargo:rerun-if-env-changed=CC
  CC = None
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  RUSTC_WRAPPER = None
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some(true)
  CARGO_CFG_TARGET_FEATURE = Some(fxsr,sse,sse2)
  cargo:rerun-if-env-changed=CFLAGS_x86_64-unknown-linux-gnu
  CFLAGS_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CFLAGS_x86_64_unknown_linux_gnu
  CFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CFLAGS
  HOST_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None
  cargo:rerun-if-env-changed=AR_x86_64-unknown-linux-gnu
  AR_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=AR_x86_64_unknown_linux_gnu
  AR_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_AR
  HOST_AR = None
  cargo:rerun-if-env-changed=AR
  AR = None
  cargo:rerun-if-env-changed=ARFLAGS_x86_64-unknown-linux-gnu
  ARFLAGS_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=ARFLAGS_x86_64_unknown_linux_gnu
  ARFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_ARFLAGS
  HOST_ARFLAGS = None
  cargo:rerun-if-env-changed=ARFLAGS
  ARFLAGS = None
  cargo:rustc-link-lib=static=lvgl
  cargo:rustc-link-search=native=/home/alix_anneraud/Git/Personnel/Xila/Core/target/debug/build/lvgl-sys-2ac71f54f14e97e9/out

  --- stderr
  `CLANG_PATH` env var set but is not a full path to an executable
  thread 'main' panicked at /home/alix_anneraud/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.69.4/lib.rs:899:13:
  assertion `left == right` failed: "x86_64-unknown-linux-gnu" "x86_64-unknown-linux-gnu"
    left: 4
   right: 8
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/std/src/panicking.rs:652:5
     1: core::panicking::panic_fmt
               at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/panicking.rs:72:14
     2: core::panicking::assert_failed_inner
               at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/panicking.rs:404:23
     3: core::panicking::assert_failed
               at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/panicking.rs:364:5
     4: bindgen::Bindings::generate
               at /home/alix_anneraud/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.69.4/lib.rs:899:13
     5: bindgen::Builder::generate
               at /home/alix_anneraud/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.69.4/lib.rs:349:9
     6: build_script_build::main
               at ./build.rs:225:20
     7: core::ops::function::FnOnce::call_once
               at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ops/function.rs:250:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
warning: build failed, waiting for other jobs to finish...
error: failed to run custom build command for `lvgl-sys v0.6.2 (/home/alix_anneraud/Git/Personnel/Xila/lv_binding_rust/lvgl-sys)`

Caused by:
  process didn't exit successfully: `/home/alix_anneraud/Git/Personnel/Xila/Core/target/debug/build/lvgl-sys-d94f608824b656ec/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=/home/alix_anneraud/Git/Personnel/Xila/Core/Modules/Graphics/include/lv_conf.h
  cargo:rerun-if-changed=/home/alix_anneraud/Git/Personnel/Xila/Core/Modules/Graphics/include/lv_drv_conf.h
  cargo:rerun-if-env-changed=LVGL_INCLUDE
  cargo:rerun-if-env-changed=LVGL_LINK
  OUT_DIR = Some(/home/alix_anneraud/Git/Personnel/Xila/Core/target/x86_64-unknown-linux-gnu/debug/build/lvgl-sys-56cb3211f74c91cb/out)
  TARGET = Some(x86_64-unknown-linux-gnu)
  OPT_LEVEL = Some(z)
  HOST = Some(x86_64-unknown-linux-gnu)
  cargo:rerun-if-env-changed=CC_x86_64-unknown-linux-gnu
  CC_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CC_x86_64_unknown_linux_gnu
  CC_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CC
  HOST_CC = None
  cargo:rerun-if-env-changed=CC
  CC = None
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  RUSTC_WRAPPER = None
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some(true)
  CARGO_CFG_TARGET_FEATURE = Some(fxsr,sse,sse2)
  cargo:rerun-if-env-changed=CFLAGS_x86_64-unknown-linux-gnu
  CFLAGS_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CFLAGS_x86_64_unknown_linux_gnu
  CFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CFLAGS
  HOST_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None
  cargo:rerun-if-env-changed=AR_x86_64-unknown-linux-gnu
  AR_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=AR_x86_64_unknown_linux_gnu
  AR_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_AR
  HOST_AR = None
  cargo:rerun-if-env-changed=AR
  AR = None
  cargo:rerun-if-env-changed=ARFLAGS_x86_64-unknown-linux-gnu
  ARFLAGS_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=ARFLAGS_x86_64_unknown_linux_gnu
  ARFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_ARFLAGS
  HOST_ARFLAGS = None
  cargo:rerun-if-env-changed=ARFLAGS
  ARFLAGS = None
  cargo:rustc-link-lib=static=lvgl
  cargo:rustc-link-search=native=/home/alix_anneraud/Git/Personnel/Xila/Core/target/x86_64-unknown-linux-gnu/debug/build/lvgl-sys-56cb3211f74c91cb/out

  --- stderr
  `CLANG_PATH` env var set but is not a full path to an executable
  thread 'main' panicked at /home/alix_anneraud/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.69.4/lib.rs:899:13:
  assertion `left == right` failed: "x86_64-unknown-linux-gnu" "x86_64-unknown-linux-gnu"
    left: 4
   right: 8
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/std/src/panicking.rs:652:5
     1: core::panicking::panic_fmt
               at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/panicking.rs:72:14
     2: core::panicking::assert_failed_inner
               at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/panicking.rs:404:23
     3: core::panicking::assert_failed
               at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/panicking.rs:364:5
     4: bindgen::Bindings::generate
               at /home/alix_anneraud/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.69.4/lib.rs:899:13
     5: bindgen::Builder::generate
               at /home/alix_anneraud/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.69.4/lib.rs:349:9
     6: build_script_build::main
               at ./build.rs:225:20
     7: core::ops::function::FnOnce::call_once
               at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ops/function.rs:250:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```
I have tried some solutions suggested [here](https://github.com/lvgl/lv_binding_rust/issues/102), which help to get rid of some error messages. However, it seems to end up being a mess because, in the case of the ESP32, the `$LIBCLANG_PATH` environment variable is set to point, which means that for the native compilation part (as used by `lvgl/build.rs`), bindgen / libclang no longer seems able to use the correct version of clang (it uses the ESP32 version instead of the native / host one). Maybe you should avoid to use lvgl-sys in lvgl build script ?

closed 2024-09-10 by AlixANNERAUD

semlanik, 2025-05-09:
@AlixANNERAUD how did you fix this issue?

semlanik, 2025-05-12:
Found the  answer, adding:
```
CROSS_COMPILE = "xtensa-esp32-elf"
```
to config.toml solved this issue.