lvgl/lv_binding_rust #184: Make this repo work again

C47D, 2024-12-19:
List of pending activities:

- [ ] Merge #165
- [ ] Merge #177 
- [ ] Merge #183 
- [ ] Discuss #176 

Sounds good @AlixANNERAUD?

AlixANNERAUD, 2024-12-19:
Yeah that seems a good start.

C47D, 2024-12-27:
Hi @AlixANNERAUD for the `lvgl-sys` move, should we discuss it in the other issue or here is ok?

AlixANNERAUD, 2024-12-27:
> Hi @AlixANNERAUD for the `lvgl-sys` move, should we discuss it in the other issue or here is ok?

In the opened issue, I think. I already left comments.

C47D, 2025-01-08:
I've added an item about cross-compilation, from the previous issues and PR's I think it stopped working somewhere. Have you tried to compile for embedded targets @AlixANNERAUD ?

C47D, 2025-01-09:
@elfaroon I saw you are interested in LVGLv9 support

AlixANNERAUD, 2025-01-14:
> I've added an item about cross-compilation, from the previous issues and PR's I think it stopped working somewhere. Have you tried to compile for embedded targets @AlixANNERAUD ?

Yes, I’ve also tried (and succeeded) in the past on ESP32 family. Libraries with C-to-Rust bindings are notoriously tricky to build, mainly due to linker and `libclang` path configurations in the build script when using `clang` and `bindgen`. Additionally, the build script for `lv_binding_rust`—particularly `lvgl_sys`—is quite complex, which further complicates matters.