lvgl/lv_binding_rust #185: Maintenance planning

kisvegabor, 2024-12-22:
Hi,

It seems the current maintainers were inactive recently and the development in this repo stalled. 

@C47D volunteered to take it over.  Thank you very much! :heart: 

Now 1 approval is required to merge PRs so we needed to either
1. remove this rule and allow any maintainers merge PRs (people can't approve their own PRs so if @C47D opens a PR he can't merge it.) 
2. Onboard one more new maintainer. It would be great to make it less lonely :slightly_smiling_face: Maybe @AlixANNERAUD? 
3. See if anyone of the current maintainers are interested in working on this repo again.

AlixANNERAUD, 2024-12-22:
I’d be happy to help, but I want to be transparent about my availability. Due to school and my professional commitments, I can’t guarantee how much time I’ll be able to dedicate to the project.

That said, I’ll do my best to contribute whenever I can!

kisvegabor, 2024-12-22:
Thank you Alix. 

> That said, I’ll do my best to contribute whenever I can!

What could it mean in the practice? Is at least 1 hour per week reasonable? If so I think it's enough to review some important PRs.

AlixANNERAUD, 2024-12-23:
I think dedicating at least 1 hour per week should be manageable for me.

kisvegabor, 2024-12-23:
Great! @C47D are you ok with inviting @AlixANNERAUD as well with write access?

C47D, 2024-12-23:
> Great! @C47D are you ok with inviting @AlixANNERAUD as well with write access? 

Yes, sounds great

kisvegabor, 2025-01-03:
Great, invitation sent!

C47D, 2025-01-08:
Can we close this issue?

kisvegabor, 2025-01-08:
Yes, thank you so much for picking up the Rust binding! :100: