lvgl/lv_binding_rust #190: [generated.rs] LvResult when C functions return void

C47D, 2025-01-10:
I was checking the generated code for `lv_label` and noticed this rust generated code returns `LvResult` when the C function returns `void`, is this expected? Do we always need to return `Ok`?

```rust
pub fn set_recolor (& mut self , en : bool) -> crate :: LvResult < () > {
  unsafe {
    lvgl_sys :: lv_label_set_recolor (self . core . raw () . as_mut () , en) ;
  }
  Ok (()) 
}
```

```c
void lv_label_set_recolor(lv_obj_t * obj, bool en);
```

clivinn-shla81092, 2025-01-10:
Hello, we tried to solve the issue.

This is what we did:

Modified the code generation logic in the `LvFunc` struct to handle void return types differently. Now, functions that return void in C will not return `LvResult<()>` in Rust, but simply `()`. This change addresses the issue of unnecessary `Ok(())` returns for void functions.

You can review changes in this commit: https://github.com/clivinn-shla81092/lvgl-lv_binding_rust-190/commit/c7cb201b9c749181ff657b79334d74c610987289.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

C47D, 2025-01-10:
@kisvegabor look at the comment above, it's AI generated :exploding_head:

C47D, 2025-01-10:
I tried the suggestion with a couple of changes and here's the output of the generated code.
```diff
diff --git a/lvgl-codegen/src/lib.rs b/lvgl-codegen/src/lib.rs
index ca97a8c..10c272f 100644
--- a/lvgl-codegen/src/lib.rs
+++ b/lvgl-codegen/src/lib.rs
@@ -130,9 +130,17 @@ impl Rusty for LvFunc {
         }
 
         // We don't deal with methods that return types yet
-        if self.ret.is_some() {
-            return Err(WrapperError::Skip);
-        }
+        let is_void_return = self.ret.is_none();
+        let return_type = if is_void_return {
+            quote!(())
+        } else {
+            quote!(crate::LvResult<()>)
+        };
+
+        // if self.ret.is_some() {
+        //     print!("Here we go");
+        //     return Err(WrapperError::Skip);
+        // }
 
         // Make sure all arguments can be generated, skip the first arg (self)!
         for arg in self.args.iter().skip(1) {
@@ -213,13 +221,20 @@ impl Rusty for LvFunc {
             });
 
         // TODO: Handle methods that return types
+        let return_value = if is_void_return {
+            quote!()
+        } else {
+            quote!(Ok(()))
+        };
+
         Ok(quote! {
-            pub fn #func_name(#args_decl) -> crate::LvResult<()> {
+            pub fn #func_name(#args_decl) -> #return_type {
                 #args_processing
                 unsafe {
                     lvgl_sys::#original_func_name(#args_call);
                 }
-                Ok(())
+
+                #return_value
             }
         })
     }
```

```rust
pub fn set_recolor (& mut self , en : bool) -> () {
  unsafe {
    lvgl_sys :: lv_label_set_recolor (self . core . raw () . as_mut () , en) ;
  }
}
```

But now we have generated code for functions that return something. They always return `Ok`.
```rust
pub fn get_recolor (& mut self) -> crate :: LvResult < () > {
  unsafe {
    lvgl_sys :: lv_label_get_recolor (self . core . raw () . as_mut ()) ;
  }
  Ok (())
}
```

Here's the original C code
```c
bool lv_label_get_recolor(const lv_obj_t * obj);
```

C47D, 2025-01-10:
With this patch we handle boolean return types (and also void)

```diff
diff --git a/lvgl-codegen/src/lib.rs b/lvgl-codegen/src/lib.rs
index ca97a8c..dc9caf5 100644
--- a/lvgl-codegen/src/lib.rs
+++ b/lvgl-codegen/src/lib.rs
@@ -129,10 +129,23 @@ impl Rusty for LvFunc {
             });
         }
 
-        // We don't deal with methods that return types yet
-        if self.ret.is_some() {
-            return Err(WrapperError::Skip);
-        }
+        // Handle return values
+        let mut has_return_value = false;
+        let return_type = match self.ret {
+            // function returns void
+            None => quote!(()),
+            // function returns something
+            _ => {
+                let literal_name = self.ret.as_ref().unwrap().literal_name.clone();
+                match literal_name {
+                    _ if literal_name == "bool" => {
+                        has_return_value = true;
+                        quote!(bool)
+                    }
+                    _ => return Err(WrapperError::Skip)
+                }
+            }
+        };
 
         // Make sure all arguments can be generated, skip the first arg (self)!
         for arg in self.args.iter().skip(1) {
@@ -213,13 +226,26 @@ impl Rusty for LvFunc {
             });
 
         // TODO: Handle methods that return types
+        let return_ok_at_the_end = if return_type.is_empty() {
+            quote!(Ok(()))
+        } else {
+            quote!()
+        };
+
+        let implicit_return = if has_return_value {
+            quote!()
+        } else {
+            quote!(;)
+        };
+
         Ok(quote! {
-            pub fn #func_name(#args_decl) -> crate::LvResult<()> {
+            pub fn #func_name(#args_decl) -> #return_type {
                 #args_processing
                 unsafe {
-                    lvgl_sys::#original_func_name(#args_call);
+                    lvgl_sys::#original_func_name(#args_call)#implicit_return
                 }
-                Ok(())
+
+                #return_ok_at_the_end
             }
         })
     }
@@ -555,11 +581,10 @@ mod test {
 
         let code = arc_set_bg_end_angle.code(&arc_widget).unwrap();
         let expected_code = quote! {
-            pub fn set_bg_end_angle(&mut self, end: u16) -> crate::LvResult<()> {
+            pub fn set_bg_end_angle(&mut self, end: u16) -> () {
                 unsafe {
                     lvgl_sys::lv_arc_set_bg_end_angle(self.core.raw().as_mut(), end);
                 }
-                Ok(())
             }
         };
 
```

Resulting rust
```rust
pub fn set_recolor (& mut self , en : bool) -> () {
  unsafe { lvgl_sys :: lv_label_set_recolor (self . core . raw () . as_mut () , en) ; }
}

pub fn get_recolor (& mut self) -> bool {
  unsafe { lvgl_sys :: lv_label_get_recolor (self . core . raw () . as_mut ()) } }
}
```

C47D, 2025-01-22:
Closed with #191

closed 2025-01-22 by C47D