lvgl/lv_binding_rust #192: Identify LVGL methods that return pointers

C47D, 2025-01-15:
> I'm not very fluent in Rust (still can't figure out ownership) but this change made `lv_label_get_text` available:
> 
> ```diff
> diff --git a/lvgl/src/widgets/label.rs b/lvgl/src/widgets/label.rs
> index 691ffb8..5235f86 100644
> --- a/lvgl/src/widgets/label.rs
> +++ b/lvgl/src/widgets/label.rs
> @@ -1,5 +1,6 @@
>  use crate::widgets::Label;
>  use crate::{LabelLongMode, NativeObject};
> +use cstr_core::CStr;
>  
>  #[cfg(feature = "alloc")]
>  mod alloc_imp {
> @@ -41,4 +42,10 @@ impl Label<'_> {
>      pub fn get_long_mode(&self) -> u8 {
>          unsafe { lvgl_sys::lv_label_get_long_mode(self.raw().as_ref()) }
>      }
> +
> +    pub fn get_text(&self) -> &'static str {
> +        let char_ptr = unsafe { lvgl_sys::lv_label_get_text(self.raw().as_ref()) };
> +        let c_str = unsafe { CStr::from_ptr(char_ptr) };
> +        c_str.to_str().unwrap_or_default()
> +    }
>  }
> ```
> 
> And I'm using it like this
> 
> ```rust
> diff --git a/examples/demo.rs b/examples/demo.rs
> index f69cf5b..dba4865 100644
> --- a/examples/demo.rs
> +++ b/examples/demo.rs
> @@ -62,6 +62,8 @@ fn main() -> Result<(), LvError> {
>      bt.set_height(80);
>      bt.set_recolor(true);
>      bt.set_align(Align::TopLeft, 0, 0);
> +    let mut test_label = Label::from("701");
> +    println!("Label get_text from bindinig: {:?}", test_label.get_text());
>  
>      let mut power: Label = "#fade2a 20%#".into();
>      power.set_recolor(true);
> diff --git a/lvgl/src/widgets/label.rs b/lvgl/src/widgets/label.rs
> index 691ffb8..5235f86 100644
> --- a/lvgl/src/widgets/label.rs
> +++ b/lvgl/src/widgets/label.rs
> ```
> 
> You can see the '701' being printed here: ![image](https://private-user-images.githubusercontent.com/11037705/401005453-7d915f50-016e-45e9-ae99-fa8ac2113c90.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4OTE4ODcsIm5iZiI6MTczNjg5MTU4NywicGF0aCI6Ii8xMTAzNzcwNS80MDEwMDU0NTMtN2Q5MTVmNTAtMDE2ZS00NWU5LWFlOTktZmE4YWMyMTEzYzkwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE0VDIxNTMwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlOWM1MTVhYTg0YmQ4NDQ3Zjk1YWI4MTMwZjY0M2NlYmU4MjY3NGM3ZTk4Zjc2ZWRlYmZhNTNlMTVhZDBmMDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ka1byTa_26e4zsyscDy3s_cq_5AWzFH0ldY6AHycBo4)
> 
> Do you think this is correct @AlixANNERAUD ?

This is a memory safety issue, since `get_text` returns a `&'static str`, as it directly points to the label's text buffer, assuming the string is valid UTF-8.
However, when the label text is modified in LVGL, LVGL likely reallocates the buffer, invalidating the pointer/reference—this behavior contradicts the `static` reference lifetime.
To address this, the function should either be marked as unsafe (with appropriate documentation) or return an owned string instead. For instance:
```rs
String::from_utf8_lossy(cstr.to_bytes()).to_string()
```

_Originally posted by @AlixANNERAUD in https://github.com/lvgl/lv_binding_rust/issues/162#issuecomment-2591193084_

C47D, 2025-01-15:
We need to identify methods that return pointers so we either document them as unsafe or handle them properly. This might be possible by using the #191 request where adding support for generation of methods that return something is being handled.

AlixANNERAUD, 2025-01-15:
Yes, that can be easily done with syn::Type::Ptr