lvgl/lv_binding_rust #193: The demos do not compile on MacOS with Apple Silicon (ARM) CPU

SketchPiece, 2025-01-15:
No matter how hard I try the demo examples do not run on macOS laptop with Apple Silicon processor

I'm using a MacBook Pro (M1 Pro) with macOS Sequoia 15.2

Before running the demo examples I update all git submodules and as stated in the documentation and other issue #35 I install sdl2 to for the simulator display to work properly on macOS
```bash
$ brew install sdl2
$ git submodule init
$ git submodule update 
```

When I'm trying to run the demo with the following command
```
DEP_LV_CONFIG_PATH=`pwd`/examples/include cargo run --example demo --features="alloc"

lvgl-sys

C47D, 2025-01-17:
Hi, just to let you know that I've read the issue and it's been added to a list of TODO items. I don't own a Mac and it might be a bit difficult for me to try replicate, but I'll try.

SketchPiece, 2025-01-17:
@C47D Maybe something I could help you with or some more info I could provide?

AlixANNERAUD, 2025-01-17:
Have you tried to install `libsdl2-dev` and not only `sdl2` ?

AlixANNERAUD, 2025-01-17:
Can you run it with [valgrind](https://github.com/jfrimmel/cargo-valgrind) and provide us the dump here ?

SketchPiece, 2025-01-19:
It looks like valgrind is not currently compatible with mac 

https://www.reddit.com/r/cs50/comments/meq16w/how_to_get_valgrind_on_macos/

When I try to install valgrind executables it throws with following error:

```
➜  ~ brew install valgrind

...

valgrind: Linux is required for this software.
Error: valgrind: An unsatisfied requirement failed this build.
```

Do you know any other software that could help?

AlixANNERAUD, 2025-01-21:
Well ... this is very surprising;
Can you try with `leaks` instead as described [here](https://www.reddit.com/r/cprogramming/comments/19148k7/valgrind_for_arm64_based_macos/?tl=fr) ?

JyCyunMe, 2025-05-18:
I had the same problem as you, then I tried your method but failed, until one time it worked.

OS: macOS Sequoia 15.4
Host: Mac Mini (2024) with Apple M4 (10)

_(Actually the version information is not the key)_
Git HEAD: d83b37407ad6c1b673c82bb9ca6cf442abac96ed
sdl2: 2.32.6

My steps:

1. Clone the repo, then you don't need to run the git submodule commands manually (it should be handled automatically)

2. Make sure you have `sdl2` installed
```sh
$ brew install sdl2
```

3. Modify `examples/include/lv_drv_conf.h`
```diff
--- a/examples/include/lv_drv_conf.h
+++ b/examples/include/lv_drv_conf.h
@@ -316,7 +316,7 @@
  *  Linux frame buffer device (/dev/fbx)
  *-----------------------------------------*/
 #ifndef USE_FBDEV
-#  define USE_FBDEV           1
+#  define USE_FBDEV           0
 #endif
```

4. ❗**Run the demo with these environment variables**❗
```sh
LIBRARY_PATH=/opt/homebrew/lib CPPFLAGS=-I/opt/homebrew/include LVGL_INCLUDE=/opt/homebrew/include DEP_LV_CONFIG_PATH=`PWD`/examples/include cargo run --example demo --features="alloc"
```

5. Now you should see the window.

[image]

I accidentally compiled successfully after a brew install something. And I discovered these diffs using `printenv`, so I think the key is, **it's missing these LIBRARY/INCLUDE path flags that includes the sdl2**
```diff
< LDFLAGS=-L/opt/homebrew/lib
< CPPFLAGS=-I/opt/homebrew/include
---
> LDFLAGS=-L/opt/homebrew/opt/ruby/lib
> CPPFLAGS=-I/opt/homebrew/opt/ruby/include
64d63
< LIBRARY_PATH=:/opt/homebrew/lib
```
@SketchPiece @C47D Hey guys, this issue should be closed now, but I suggest you update the README file with reference to this instruction.